Ultimaker/Cura #3491: New feature, allow pause and removal of partial multi part print

cjbeckjr, 2018-03-14:
**Application Version** all versions

**Platform** all platforms

**Qt**

**PyQt**

**Display Driver**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results** would like to be able to run multi part print and if a portion fails be able to remove that failed portion from the remaining print by either reslicing and restarting at a certain layer, layer of pause would need to be logged to restart

**Additional Information** this seems to be a feature that might be fairly easy to implement provided it is run from the application and not through SD card

ianpaschal, 2018-03-15:
I think there's two parts to this, right?

1. Pause print after an object is completed.
2. Re-do that object if needed.

Part 1 you can already do quite easily on an Ultimaker machine and on 3rd party machines that's a bit beyond our control.

For part 2, that would require jumping around within the gcode file which again is a bit beyond our control. Once it's on the printer it's the printer's firmware which is reading the gcode and doing stuff with it. So on a networked UM3 machine it would be possible to implement but I think _highly_ unlikely, and again, on a 3rd party machine: beyond our control.

I will see if someone else wants to give their opinion on this but I'm pretty sure this is a `Won't Do`.

As for your workflow, I would just use option 1 and then re-run a new print job with failed parts.

cjbeckjr, 2018-03-15:
This is during the print not after.
If printing a multiple item print and 1 of the many fail currently the
print continues and potentially causes spaghetti ruining the entire print.

Requesting pause and delete that lost object from the print and be able to
restart the remaining objects on the print without that lost object.

If monitoring remote knows which layer it being printed, this could be
saved at pause from the application.
Objects left on the build plate and print restarted at the last layer.

This would mean allowing print in air and slice in air

ianpaschal, 2018-03-15:
> This is during the print not after. If printing a multiple item print and 1 of the many fail currently the
print continues and potentially causes spaghetti ruining the entire print.

I thought you were talking about one-at-a-time mode. It's true that removing the object underneath will cause spaghetti but then you should just restart the print because...

> This would mean allowing print in air

...that's physically impossible. Printing in the air will always cause spaghetti because there's nothing for the print to attach to.

Unless of course you re-do the failed object but this is:

- Maybe possible in one-at-a-time mode, but in that case one failed object will not affect the others anyway, but;
- Not possible in normal mode as you'd likely either crash into existing parts or print spaghetti, plus there's no way to know what line is part of what object. As far as the printer knows, they're just lines in space.

Also does this mean your "multi-part print" is stacked on top of each other? In that case, don't stack it! Lay out each part and print them in one-at-a-time mode and even if one fails the others will not and you can just re-do the one that failed next time.

Since this is something between a non-issue and not-technically possible, I'm going to close this. Sorry about that.

closed 2018-03-15 by ianpaschal

Ghostkeeper, 2018-04-13:
I think there's a bit of a misunderstanding going on here. Cjbeckjr means printing two objects normally, all-at-once. If one object starts to fail and produce spaghetti, the spaghetti often ends up ruining the second object.

You can currently pause from Cura (as long as Cura is connected to the printer). But you can't then remove part of the print while paused and continue without the failed part. That requires changing the store of g-code on the fly while the printer is printing it. Cura is able to do that while printing via USB, but not while printing via wifi or on a USB stick because the g-code is in the printer then.

cjbeckjr, 2018-04-13:
Ghostkeeper you are correct, also Prontoprint has this feature and we have
subsequently switched to running the prints from there.