Ultimaker/Cura #3493: enabeling, disabeling extruders

maukcc, 2018-03-14:
Great feature ! Thanks a lot.
To make it fantastic:
- when you right click an extruder you get: en/dis-able extruder, can you make that en/dis-able <variant_ name> ?
- Can you grey out disabled extruders even more. The difference between the black and the grey is to little, so it is not clear in one glance that the disabled extruders are in fact disabled.
- Can you grey out the color circle as well. (or loose it all together when the extruder is disabled)
- If one clicks on a disabled extruder we would expect a popup saying "This <variant_name> has been disabled, please enable it to make any changes", and nothing further would happen.
- is there a way to disable extruders by default in the definition file or extruder.def.json file?

Ghostkeeper, 2018-04-12:
> when you right click an extruder you get: en/dis-able extruder, can you make that en/dis-able <variant_ name> ?

I don't really agree with this. You're disabling the extruder train, not just the nozzle. The feeder also stops, etc.

> Can you grey out disabled extruders even more. The difference between the black and the grey is to little, so it is not clear in one glance that the disabled extruders are in fact disabled.

Yeah it is a bit unclear. We've held with the 'disabled' colour that the rest of the theme uses though and removed the outline to make it even more unclear. Do you also think it's unclear when you override all child settings of a parent setting and the parent setting gets disabled?

> Can you grey out the color circle as well. (or loose it all together when the extruder is disabled)

Yeah I think removing it would be nice! @LukeChen-Ultimaker?

> If one clicks on a disabled extruder we would expect a popup saying "This <variant_name> has been disabled, please enable it to make any changes", and nothing further would happen.

That doesn't behave like a disabled button though. And giving too many pop-ups is annoying.

> is there a way to disable extruders by default in the definition file or extruder.def.json file?

Not currently. The preferred method is to provide definitions for as many extruders as maximally supported and then change the number of extruders in the Machine Settings dialogue. Here we make the distinction between modifying the printer itself (changing the number of extruders) and preventing settings from other extruders to influence the globals and allowed print volume (disabling extruders).

> When you place an object on the buildplate, you can choose a disabled extruder to print it with(on the left side of the screen, without getting an error. Maybe let the disable extruders disappear on the left side of the screen?

Yeah that sounds like a bug to me, but I can't reproduce it. The button on the left hand side is disabled. Maybe that was fixed since then?

> Can we have an option(because not everyone will like this option), where you start Cura, all extruders are disabled by default, then you place an object and you have to choose with which extruder you want to print it with, and that extruder gets enabled. So every time you place an object you get a popup asking for which extruder to use. In short, if an extruder is not used for any object on the build plate it is disabled by default.

Currently you can't disable the last extruder any more. You need one extruder at the least. This is partly because it doesn't make sense to have a print that doesn't use any extruders, and partly because it was giving bugs in our code to change the Support Extruder and such when the current support extruder gets disabled. Also, again, giving pop-ups is annoying if you do that too much.

maukcc, 2018-04-13:
> I don't really agree with this. You're disabling the extruder train, not just the nozzle. The feeder also stops, etc.

What we are trying to do, is that variant is not just the nozzle but it changes from extruder to laser or drill or...  
So then it reads a bit silly if it says extruder. No biggie.