Ultimaker/Cura #3494: [Master] UM2+ material with no quality

ChrisRiddell, 2018-03-14:
**Application Version**
Master as of (8efc3b8b5d454d5b5bae8687da8740879d444c48)

**Platform**
MacOS 10.13

**Qt**
No idea hahaha

**PyQt**
PyQt_5.8.2

**Printer**
Ultimaker 2+

**Steps to Reproduce**
Click the drop down you will see a material that was not their in 3.x and that's PETG

**Expected results**
I don't think that should be there as there is no quality for it in `resources/quality/ultimaker2_plus/
`

ianpaschal, 2018-03-15:
Yesterday we fixed some issues relating to materials and quality. Can you provide a more detailed explanation of how to reproduce?

ChrisRiddell, 2018-03-15:
in the photo below you can see there is two materials in the dropdown that where not their in 3.2< and their is no quality files for them in the folder resources/quality/ultimaker2_plus/

[image]

ianpaschal, 2018-03-15:
Ah those! Yes that's to be expected. You can feel free to make a quality file for them if you like but it was mentioned in stand up and our product owner said we should keep them as is for now.

closed 2018-03-15 by ianpaschal

ChrisRiddell, 2018-03-15:
Okay after looking at some of the code and the FDM_Materials repo this now makes sense.