Ultimaker/Cura #3495: font problems on Mac OS High Sierra 10.13.3

aupilot, 2018-03-15:
[image]

fieldOfView, 2018-03-15:
See #3277

aupilot, 2018-03-15:
thanks. Is there a binary with these fixes available?

ianpaschal, 2018-03-15:
There's not. The download links in those threads were mainly just meant for testing. We didn't really want to distribute that because it makes a headache down the road when we say, "What version are you running," and a user says, "Uh, something I downloaded from a GitHub issue thread."

But since we know @fieldOfView's fixes work and they're incorporated into our development versions, they'll be in 3.3 Beta in about 2 weeks.

closed 2018-03-15 by ianpaschal