Ultimaker/Cura #3497: Increase shell thickness / infill density near sharp angle

stibinator, 2018-03-15:
I am printing an object that has several thin struts coming out of a flat plate. Initial testing of a print showed that at normal print settings the struts tend to snap off at the junction with the plate, shearing along the x/y plane (i.e. parallel to the print bed). In this case the plate is also parallel to the bed, so the struts vbasicall break off at the layer junction at their base.

This is because there is a concentration of stress at the sharp junction, and an inherent weakness along the x/y plane in FFF objects. While it can be fixed with better design of the object, or increasing settings like infill density or shell thickness for the model overall, it would be much better if reinforcement could be added at sharp angles locally.

So what I propose is that shell thickness and infill density could be increased according to the distance from an acute angle in the shell. It could be like the Gradual infill steps, but instead of being based on distance from the top it would be distance from an acute angle. 

There could be a threshold above which an angle is deemed acute, or there could be a continuum so the effect would get stronger as the angle got more acute.

The result would be a little like welded steel pies, where the weld acts as an extra reinforcement around a join, except it would be on the inside. Since it would be based on the radius from the angle it would form a circular bead around the join, reducing the concentration of stress at these points.

Attached is a screenshot of the current print, and a mockup of what the internal fillet would look like (in blue, the exterior geometry is yellow).

[image]
[image]
[image]
[image]
[image]

BagelOrb, 2018-03-16:
Your idea is very interesting.

However your thinking is based very much on your specific model. When you think about a small human figurine for example you'll see that sharp corners occur everywhere due to the great amount of detail in the model.

Ideally the stress in those regions is identified and tackled head on using some form of topology optimization for the infill. This is very much a contemporary field of research, which I hope to commit to in the coming years. It will take a while for a proper solution to be implemented.

Meanwhile you can edit the density yourself by loading those blue regions models and making them an Infill Mesh. Then you need to tweak all settings like infill percentage to 75%, wall thickness and top bottom thickness to 0 for that mesh only.

stibinator, 2018-03-16:
That's a good point, I guess the analysis of the mesh is more complicated than it first appears. I didn't knopw about infill meshes, I'll go and research them now.