Ultimaker/Cura #3498: Bad interface colours causing me fatigue and sending me colourblind.

adamantiz, 2018-03-15:
I, like many others, find it extremely difficult to view white text on a darker background.

In your main colour scheme you have light blue info text dialogues with white text on them.

In your dark colour scheme you have dark interface with white text.

Looking at your program for even a short time - starts to burn my eyes and I have to walk away.

Please amend this issue, and give a NORMAL standard colour scheme, a white screen with black text. Info dialogues with black text etc...

ianpaschal, 2018-03-15:
Thanks for your feedback.

Unfortunately, I really don't understand it. The default color scheme _is_ a white screen with black text. If you can't find it, go to Settings > General > Theme and select Ultimaker vs. Ultimaker Dark.

Also what is awful on your eyes is sweet relief on other people's eyes. That's why my code editor is _always_ light text on a dark background. Can't imagine staring at black text on white all day. Yuck. 😉

closed 2018-03-15 by ianpaschal

fieldOfView, 2018-03-15:
Fortunately you can easily change colors with a custom theme.

* In Cura, go to Help -> Show configuration folder.
* In the folder that opens, navigate to the "themes" subfolder. 
* Create a new folder named eg "grey_tooltips", and place the following file in it:
https://gist.githubusercontent.com/fieldOfView/35091a3278cb070b005bf7c25e1c5817/raw/c55d7f1055f7bacb86d449bc5489182d2da9e160/theme.json
* Restart Cura, go to Preferences and select the theme named "Grey tooltips"
* Restart Cura again

You should now have tooltips that have black text on a medium grey background.

Ghostkeeper, 2018-04-13:
> The default color scheme is a white screen with black text.

Except the tooltips, which have white text on a blue background.