Ultimaker/Cura #3500: master branch setting menus all blank

smartavionics, 2018-03-15:
**Application Version**

master branch

**Platform**

Linux

**Qt**

5.10.1

**PyQt**

5.10

**Display Driver**

2018-03-15 18:51:50,393 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [93]: OpenGL Version:  4.5 (Core Profile) Mesa 17.2.4
2018-03-15 18:51:50,393 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [94]: OpenGL Vendor:   Intel Open Source Technology Center
2018-03-15 18:51:50,393 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [95]: OpenGL Renderer: Mesa DRI Intel(R) HD Graphics 520 (Skylake GT2)
**Printer**

**Steps to Reproduce**

Open any drop-down setting menu

**Actual Results**

All of the menu entries are blank

**Expected results**

The menu entries should be visible.

**Additional Information**

The selected value is displayed correctly but all the drop-down options are blank.

I recently updated my installation using the latest cura-build-environment so maybe this is related to the change in qt pyqt versions?

fieldOfView, 2018-03-15:
Screenshot?

smartavionics, 2018-03-15:
Sorry, no. When I grab the screen, the menu closes.

ianpaschal, 2018-03-16:
PyQt does sound like the most likely culprit for this kind of thing. Since we got all kinds of weirdness in 5.10.0 I'd still advise running with 5.8.2. That's what we're using internally for development.

My guess is that should return things to normal.

smartavionics, 2018-03-16:
OK, but I'm just using the master branch of cura-build-environment. If that isn't good, please tell me which tag/commit in that repo I should be using. Thanks.

smartavionics, 2018-03-16:
I'm just going to revert back to what I was using a week ago and see if it recovers...

ianpaschal, 2018-03-16:
Really? That's strange. I was under the impression that the cura-build-env repo had been reverted to use 5.8.2 by @LipuFei

smartavionics, 2018-03-16:
No, it's still using 5.10.1, he didn't revert...

ianpaschal, 2018-03-16:
I was just looking at the repo's commits which I guess I should have done before writing that comment.

It's possible that something was broken in Chris's work but it seems more likely that it's related to the PyQt version. Do you know what was the last commit you were using that it worked?

smartavionics, 2018-03-16:
Not 100% sure, I possibly have been using a9ae4b.

ianpaschal, 2018-03-16:
I'm not able to find that commit. But I'd say try reverting to a pre-5.10 commit and see if that fixes it. If not, we'll take another look and possibly see if @ChrisTerBeke knows of some change he made that could affect menus.

ChrisTerBeke, 2018-03-16:
I haven't touched the PyQt version.

smartavionics, 2018-03-16:
I have reverted back to https://github.com/Ultimaker/cura-build-environment/commit/a9ae4b2efe9f89772f1c71b7245606857f62870c (currently building) and will let you know if that works...

LipuFei, 2018-03-16:
@smartavionics It was caused by the setting visibility preset menu. It was fixed yesterday. Could you try again? If it still doesn't work, try removing `general/visible_settings` field in `cura.cfg`.

smartavionics, 2018-03-16:
Oh, really, well I've now started replacing my cura environment so I will have to let that run to completion (takes a while) and will let you all know what happens later. Cheers.

ianpaschal, 2018-03-16:
> I haven't touched the PyQt version.

I know. That's why I said that if switching back from PyQt 5.10 doesn't fix it, maybe it was something else. 😜

smartavionics, 2018-03-16:
OK, I just use the latest master branch Cura with the latest build environment and the menus are still blank. Removing general/visible_settings didn't make any difference. Will let you know if it works again when the reverted build environment has been installed.

fieldOfView, 2018-03-16:
> It was caused by the setting visibility preset menu.
[image]

What did I break?

LipuFei, 2018-03-16:
@fieldOfView No worries, it was fixed. Something went wrong with the custom visibility storing/restoring, which can make all settings go away...

smartavionics, 2018-03-16:
As I said a few comments back, the menus are still blank with the latest master branch so I think it is more likely the build environment that is causing the problem.

fieldOfView, 2018-03-16:
Feel free to tag me in commits that fix errors caused by things I added; I would like to learn from my mistakes too.

LipuFei, 2018-03-16:
@fieldOfView Tagged for you in 8e39849

fieldOfView, 2018-03-16:
Note that @smartavionics is talking about dropdown menus, not the custom settings in the sidebar.

ianpaschal, 2018-03-16:
@fieldOfView He mentioned "settings" drop downs. Also that the selected setting displayed fine, but that the drop down under it was blank. This is what lead to me blame QML/PyQt 5.10

LipuFei, 2018-03-16:
That's weird then... I've been using PyQt 5.10 + Qt 5.10 for a long time and things seem fine to me.

LipuFei, 2018-03-16:
With PyQt 5.10.1, it always segfaults for me at launch, so I don't even have a chance to see the dropdown menus.

ianpaschal, 2018-03-16:
@LipuFei Yeah I use 5.10 a lot but every once in a while I have some QML explosion and it can usually be traced back to not using 5.8.2.

fieldOfView, 2018-03-16:
> He mentioned "settings" drop downs
[image]

These, right?

smartavionics, 2018-03-16:
No, the drop down menus in the panel on the right, stuff like "Infill pattern", "Z Seam Alignment", etc.

BTW I have now reverted back to my previous build environment which must be quite old as it's using qt 5.8.0 and the menus are now OK again.

ianpaschal, 2018-03-16:
Cha-ching! I'm still really proud when I correctly diagnose a Cura problem.

We're keeping a close eye on when we can upgrade Qt/PyQt but since both 5.10 and 5.10.1 are causing problems I'm not sure when that would be.

ianpaschal, 2018-03-16:
Also in this case I'm going to call this one "fixed" as using 5.8.x is the best possible solution right now.

closed 2018-03-16 by ianpaschal