Ultimaker/Cura #3501: Improviment on Cooling x bridge

mtmello2003, 2018-03-15:
Just a suggestion:
Option to turn on fan before does a bridge.
Something like:
"Fan on before bridge"
    "Fan bridge max speed"
    "Fan on before 'n' bridge mm'
    "Fan off after 'n' bridge layer"
    "Fan off after 'n' bridge mm"

Sometimes the object printed does not need cooling at all but the bridges gets better quality with fan turned on...

thanks

ozfunghi, 2018-03-20:
I made a feature request last month, that will be included in 3.3. It will let you set slow down a print for overhangs (x% speed at x° overhang), to get better results, without curling up or needing supports. Slower print means better cooling as well. Maybe they could include fan control at the same time.

ianpaschal, 2018-03-27:
There are some improvements to bridging in 3.3 that I think will satisfy your needs.

closed 2018-03-27 by ianpaschal

filipgoc, 2018-04-18:
> There are some improvements to bridging in 3.3 that I think will satisfy your needs.

Looking forward to putting that to the test!