Ultimaker/Cura #3502: Improvements on "Z Hop when retract"

mtmello2003, 2018-03-15:
Suggestion: adds the option to:

"Z hop start layer"
"Z hop stop layer"

Then, this function will run only between the start/stop user defined layers.

Some design to be printed needs it in specific areas to improve quality, not at the entire model...

BagelOrb, 2018-03-16:
There are loads of features which might be optimal on some layers, but suboptimal on others. Having separate settings for all those features to turn them on and off wouldnt be the right way to go about.

I don't think ultimaker is ever going to put time in fixing this.

closed 2018-03-16 by BagelOrb

fieldOfView, 2018-03-16:
> I don't think ultimaker is ever going to put time in fixing this.

I would put it in this way: "I don't think ultimaker is ever going to put time implementing the suggested fix."
It is something that could be implemented as a post processing plugin.

mtmello2003, 2018-03-16:
Just a thing to consideration: the Z hop feature can stress a lot the Z axis motor then, I think this features I suggested could be relevant against some others.
Tks