Ultimaker/Cura #3503: Problem adding new materials, i believe the cause is the filament dia.

RacerBo, 2018-03-15:
Like some of the other users, i also have problem adding new material to Cura 3.2.1
I previously created this material in v3.2 and exported it out.
With 3.2.1 it will not import.
I try to create new material in V3.2.1.
All the new setting will stay (exited out, restart..)
until i type in the 1.75mm dia filament.

ianpaschal, 2018-03-16:
Please use the provided template when you create a new issue. We need a lot of info here like your logs and also a more detailed description of what is going on. When you say "it will not import" what happens? Nothing? Crash? Error pop up?

RacerBo, 2018-03-16:
**Application Version**
3.2.1

**Platform**
windows 10 x64 home

**Qt**

**PyQt**

**Display Driver**
Gigabyte Radeon RX 550 D5 2GB , AMD Driver [17.8.1]

**Printer**
custom UMo+ clone

**Steps to Reproduce**
(1st attempt)
 I previously created a PETG material profile (Inland brand, green color, 238 temp, 1.75mm filament...) in v3.2. Worked great, not a problem. After a fresh install of Windows 10x64, and a fresh install of v3.2.1 (new computer), i then try to import the same material profile.
(2nd attempt)
I then try to create the same new material in V3.2.1.  I duplicated the generic PETG profile, unlink the material and edit the specs , 1 at a time to test if the settings will stay. After each edit, i verify that the new profile still shows in the material list page.

**Actual Results**
(1st attempt)
 nothing happen. no error message, no crash. the program pop back to the material page. 
The new material did not import.
(2nd attempt) 
All the new setting will stay, brand name, color, temperature, at **2.85mm** dia ... This material profile stayed even after exiting Cura.
But, when i edit the material profile to type in the 1.75mm dia filament, the entire profile disappear from the list.

**Expected results**
 (1st attempt)the new material should have imported.
(2nd attempt) the new profile should have stayed.

**Additional Information**

Ghostkeeper, 2018-04-16:
I can't reproduce this issue. I exported a material in 3.2.0 and imported it in the current state of 3.3. Could you try this with the 3.3 beta and see if it still occurs?

RacerBo, 2018-04-16:
Hi, I figured out the issue. The printer profile must have the the 1.75mm setting active and in use.
If i do not have a printer with an filament setting of 1.75 mm profile active, i can not create a material profile of 1.75 mm or import a material profile. thanks for your help.

closed 2018-04-16 by RacerBo