Ultimaker/Cura #3504: Modifying Materials Causes the Original Materials to be Modified as Well

fatalelieberi, 2018-03-16:
**Application Version**
Cura 3.2.1

**Platform**
Windows 10 Pro 64bit Version Version	10.0.16299 Build 16299

**Qt**
LGPLv3

**PyQt**
GPL

**Display Driver**
nvldumdx.dll Version 23.21.13.8813

**Printer**
Tevo Tornado
[CurrentCuraProject.curaproject.zip](https://github.com/Ultimaker/Cura/files/1817497/CurrentCuraProject.curaproject.zip)

**Steps to Reproduce**
Details can be viewed in the attached video in the Zip in Additional Information section. But, to summarize:
1. Open Manage Materials window
2. Modify existing but duplicated material and unlink the material
3. Close Cura
4. Open Cura
5. Check for Materials in the list

**Actual Results**
The original material that was duplicated from got modified as well, although only the information tab that got changed

**Expected results**
The original material shouldn't get changed

**Additional Information**
[Cura Bug when Modifying Materials.zip](https://github.com/Ultimaker/Cura/files/1817500/Cura.Bug.when.Modifying.Materials.zip)

ianpaschal, 2018-03-27:
Fixed in 3.3.

fatalelieberi, 2018-03-27:
I've tried in 3.3 but the issue still persists, although randomly. At one time, it opens correctly, but at the other time, my edits are changed, usually when I modify brand. The original material will get moved to the duplicated material's brand. I've unlinked the materials also. Maybe it's because 3.2.1 still installed in my system? I'll try to uninstall it to confirm it.

Ghostkeeper, 2018-04-20:
Also not able to reproduce this right now in 3.3. I don't think anything was changed though since Ianpaschal tested this 24 days ago. Were you able to confirm it again, @fatalelieberi?

jackha, 2018-05-08:
In our master version we added some extra error checking and it seems there is something wrong with the profile generic_petg_175_cartesio_0.25mm_thermoplastic_extruder. We hope that we get new ones soon and that it will fix the issue. Using other materials in 3.3.1 or master should work fine.

closed 2018-05-08 by jackha