Ultimaker/Cura #3509: Feature request

DavidMnDenver, 2018-03-16:
**Application Version**

3.2.1
**Platform**

10.13.3 (High Sierra)

It would be nice if Cura would light each completed layer in the display as the project prints. This would give a visual representation of the project's progress...

Ghostkeeper, 2018-04-16:
And if the user has been slicing something else in the meanwhile, while it's printing? Or if the user didn't slice it at all but is just connected to a printer that is printing something that someone else is printing?

This involves sort of tracking whether the currently displaying layer view is the print that is currently printing on the connected printer. When printing via USB this is possible, but any other connected printers, such as Octoprint or UM3's network functionality, won't give us the guarantee that the print that just started on the printer is the same as the one that we just sent.

VanessaE, 2018-05-13:
Just fwiw, Pronterface goes a step farther: it highlights the segment it's printing (well, it'll always be several segment ahead of the printer, natch).

I'd like to see this added to Cura - why not re-use the existing layer "playback" feature (where it shows a ghost of the hotend)?

Of course that leaves the question of slice-while-printing unanswered...

ruddy17, 2019-05-19:
I am using cura to send the print to octoprint, which then prints it on my non'makerbot printer. Right now I prefer monitoring the progress in octoprint web interface, but I feel it is not the most convenient imaginable way. I would like cura to add some kind of listening port, that listens to gcode lines proxied from octoprint and sets layer progress live in preview pane. This way I would not have to switch to browser and eliminate one more redundant interface along the workflow, which is an obvious advantage UI/UX'wise. Also I believe it should not be hard to implement for anybody familiar with cura codebase.

Regarding the slice-while-printing: the live monitoring feature may be enabled from Monitor page, so whenever one clicks Slice, monitoring mode disables and normal slicing runs, but whenever someone wants to return to live monitoring, he just clicks the button

closed 2020-06-20 by no-response[bot]