Ultimaker/Cura #3510: Printing Profiles are not chnaged after change nozzle

GentleRV, 2018-03-17:
**Application Version**
 3.2.1

**Platform**

**Qt**

**PyQt**

**Display Driver**

**Printer**
Prusa i3 MK2

**Steps to Reproduce**
 change nozzle diameter in machine settings

**Actual Results**
nothing

**Expected results**
change of first layer thickness and all wall thicknesses with a multiple of the nozzle diameter above a minimum 

**Additional Information**

GentleRV, 2018-03-17:
Many things are so called calculated in Cura, but when the nozzle size is altered profiles stay the same

closed 2018-03-17 by GentleRV

Ghostkeeper, 2018-04-16:
Not able to reproduce this bug. The line width changes properly.

The first layer thickness and wall thickness is not supposed to change with the nozzle size. Well, maybe the first layer thickness should but it's currently not expected to as we have no formula to compute this relation.