Ultimaker/Cura #3512: Printer "Machine Settings" dialog box layer order gets confused when you leave/return from Cura

pbradstr, 2018-03-17:
The **Settings:Printers:Profile** menu command brings up a **Preferences** dialog box. Then if you select *Printers* in that box, it gives you another button in that dialog, **Machine Settings**. that brings up a 2nd level child dialog box. 

This issues is that external actions to cura can change the expected default **stay-in-front** ordering on this second *Machine Settings* dialog box, allowing you unknowingly **hide** first **Preferences** dialog box **IF** you do the following steps:

1. From Cura, You go out it by selecting some non-Cura application on your desktop - like Firefox or the Terminal window. 
2. Then come back into Cura by selecting the parent **Preferences** dialog - it comes up and obscures the child **Machine Settings** dialog box. (You may have to do the 1 --> 2 punch a couple of times)*
3. With the order now wrong, the **Close** button on the **Preferences** dialog box doesn't work *(and it's not grayed out as it should be)* because it's child **Machine Settings** dialog is not closed first*

You can recover/cycle through this order change as may times as you like, this by repeating step 1 again, and then select the in-the-back child **Machine Settings** dialog, which will pop it in front. 50% of the time, this will be wrong. 

**NOTE:** This same "out of order" window issue also occurs with the **Upgrade Firmware(* dialog window, also under **Preferences/Printers**

**Application Version**
Cura 3.2.1

**Platform**
Macintosh Macbook Pro, MacOS Version 10.13.1 High Sierra

**Qt**
From website

**PyQt**
From website

**Display Driver**

**Printer**
Not applicable  - (but I was using a  Select Mini V2 (E3D)) 

**Steps to Reproduce**
See description above

**Actual Results**
Lose window,  **Close** button on dialog box doesn't work, and it makes no sense why

**Expected results**
Should never be allowed to put a child window behind it's parent window

**Additional Information**

Below I've pulled the **Machine Settings** dialog box out to the side to show it -- by default, it's centered directly behind the main **Preferences** dialog

diegopradogesto, 2018-03-20:
We tested in different OS and it works as expected. If you have the "Machine Settings" dialog opened then the "Preferences" window is always behind the dialog, even changing between applications.
Let's wait for other people to confirm that.

pbradstr, 2018-03-20:
Except when it isn’t.

I was on a Mac, 100% repeatable. 

Yes, this is totally unexpected behavior, that would be dependent upon the API layers on top of the window management software.  What were you testing on? 

Peter Bradstreet
(408) 499-0106

diegopradogesto, 2018-03-20:
I tested on Mac and Windows trying to follow the steps you indicate and I couldn't reproduce it. Is it possible for you to upload a small video?

pbradstr, 2018-03-20:
Video is Good idea, will do in next couple of days. Thanks

Peter Bradstreet

Ghostkeeper, 2018-04-16:
Not reproducible on Ubuntu 17.10.

A video will serve as "proof" that it happens, but it won't allow us to make a fix if we can't reproduce it.

closed 2018-05-08 by jackha