Ultimaker/Cura #3513: Internal hole cover is printed from the middle out (in the air)

aronsky, 2018-03-17:
**Application Version**
3.2.1

**Platform**
macOS

**Printer**
Creality CR10 Mini (Regular CR10 profile with decreased bed size)

**Steps to Reproduce**
Create a model with a pocket for a screw nut. Make a hole through the pocket (the hole the screw will go through). See the attached file. The nut is supposed to be put into the printed object during printing (the printing should pause at the beginning of the layer that covers the nut, to allow the insertion of the nut).

**Actual Results**
The cover for the nut is printed in the air (from the inside out).

**Expected results**
The cover for the nut should be printed from outside in, to avoid printing in the air.

**Additional Information**
Notice the last line printed by the extruder in the following screenshot: it has no supports underneath, and originates in thin air:
[image]

See layer 18 in the following project:
[Cura Bug.zip](https://github.com/Ultimaker/Cura/files/1821457/Cura.Bug.zip)

If you examine the STL, the problem occurs at layer 18 when slicing it at 0.2mm, but repeats itself at other layer heights as well (simply at different layer number).

smartavionics, 2018-03-17:
Cura is working as intended. The normal technique used in this situation is to place a solid layer which is just 1 layer thick at the top of the nut trap so that the walls that are above it are printed on plastic and not air. Here's an example of what I mean:

[image]

[image]

[image]

aronsky, 2018-03-17:
if I understand correctly, the screw would then have to pierce that single layer later on?

If so, it definitely doesn't seem like Cura is working as intended - on the contrary, what you suggest looks like a workaround for a known erroneous behavior...

smartavionics, 2018-03-17:
Yes, you have to drill out the single layer once the print has completed.

I very much doubt that Cura will behave as you imagine it should any time soon. Sorry.

aronsky, 2018-03-17:
Is there no way to reorder the extrusion in the layer so that it starts from the outside perimeter of the nut cap towards the inside? Maybe add some sort of experimental setting for such cases?

Usually, Cura has no trouble creating small overhangs (and if they are small indeed, they turn out fine in print). This seems like a type of an overhang - why would Cura not detect that it's extruding material into thin air, without any connection to the previously printed part?

diegopradogesto, 2018-03-20:
What happen here is that the walls are printed always before the skin (bottom in this case). Even changing the "Infill before walls" it's something that doesn't change. The trick that @smartavionics comment is probably the best solution in this case.
For this special case we could probably implement a sort of "Skin before walls" setting. However at this moment, it is not a priority and we are not doing it, since you can do some workaround for that.

RRAlex, 2018-08-31:
I've just hit the same issue for the first time on 3.4.1, instead of curving the top from the wall bellow, it starts in the center.
None of the suggested infill options fixes it though, so I guess I'll go with the single layer ideal, but that's going to result in a messy print.
[image]

mazamorac, 2019-11-30:
I'm convinced that implementing the "Skin Before Walls" option would fix **most** overhang issues that can't be done with square bridging (as described in the [workaround ](https://github.com/Ultimaker/Cura/issues/3513#issuecomment-373933210) above in the issue thread)

If you also add heuristics to start printing adjacent to previously printed lines (either from the previous layer or the current layer, you'd eliminate the need of the straight bridge workaround.