Ultimaker/Cura #3517: [3.2.1] Crash when attempting to change material diameter in machine settings

ewortzman, 2018-03-17:
**Application Version**
3.2.1

**Platform**
Windows 10

**Printer**
An custom machine was selected

**Steps to Reproduce**
Attempt to change the material diameter in machine settings

**Actual Results**
Cura immediately crashes

**Expected results**
The material diameter setting should be saved and Cura should not crash

ChrisTerBeke, 2018-03-18:
Please post the logs here, without them we can't even begin to guess what happened.

smokey1300, 2018-04-12:
Same problem.

ianpaschal, 2018-04-12:
In the words (pictures?) of @fieldOfView:

[image]
[image]
[image]
[image]

ChrisTerBeke, 2018-05-23:
No new information appeared. Closing.

closed 2018-05-23 by ChrisTerBeke