Ultimaker/Cura #3518: [3.2.1] changing material diameter when creating a custom material causes material to disappear

ewortzman, 2018-03-17:
**Application Version**
3.2.1 

**Platform**
Windows 10

**Printer**
An machine selection

**Steps to Reproduce**
1. Create a new custom material, or duplicate an existing material
2. Attempt to change the material diameter

**Actual Results**
The material immediately disappears from Cura and cannot be retrieved

**Expected results**
The new diameter setting should be saved and the material should not disappear

**Additional Information**
According to Issue #3283, This bug was fixed in version 3.2.1.  However, I still experience it.  I have confirmed that I am, in fact, running version 3.2.1

t-paul, 2018-03-18:
Confirmed on Debian/Testing with the AppImage 3.2.1.

Activating the material sometimes seems to accept and store some of the values so the material appears in the list. But even then it does randomly lose settings, e.g. changing temperature, tab to next setting, close dialog will just discard the setting and revert to the previous temperature.

ChrisTerBeke, 2018-03-18:
We only show materials that fit the current machine selection. If you have a 2.85mm extruder selected and change the material to be 1.75 (or visa versa) it'll indeed be removed from the materials list. Try changing the extruder filament diameter to match the material diameter, then it should appear again.

I'm not saying this is the best UI/UX, but this is how it currently works and that's expected. @LukeChen-Ultimaker any ideas to improve this in the future?

ewortzman, 2018-03-19:
I can understand materials that are not compatible not showing up in the main view.  But when in settings and creating a new material or modifying an existing one, all materials should be present.

RacerBo, 2018-03-20:
Same issue here. 
Per my issue report  #3503, i notice it only happens after the dia.(1.75mm) is inputted.
Otherwise, all the other new material settings will stick. 
Also, my my custom umo+ (nozzle 1.75mm) printer was selected when the new materials are being created.

LukeChen-Ultimaker, 2018-03-27:
A way we can handle this:
1. Once user input a different diameter then show a gentle warnings to remind the users that the diameter does not match with extruder.
2. The text will state something like "Material diameter does not match extruder diameter, please change extruder extruder at machine settings".
3. The duplicated material profile should remain and keep the original diameter value.

Ghostkeeper, 2018-04-20:
We already have a warning like that:
[image]

Ghostkeeper, 2018-04-20:
We've chosen not to display materials with different diameters on purpose, because then half of the materials are not applicable to your current extruder. Most people have only one type of printer so for most people it would be unnecessary.

closed 2018-04-20 by Ghostkeeper