Ultimaker/Cura #3519: crashes on loading STL file

biscuitlad, 2018-03-18:
**Application Version**

3.2.1
**Platform**

ubuntu bionic
**Qt**

?
**PyQt**

?
**Display Driver**

?
**Printer**

K8400 Dual nozzle
**Steps to Reproduce**

Load STL - see link to file below
**Actual Results**

CuraEngineBackend.CuraEngineBackend.slice [197]: starting to slice!
2018-03-18 17:16:32,636 - DEBUG - [(131717719598912)-MainThread] CuraEngineBackend.CuraEngineBackend.slice [214]: Going to slice build plate [0]!
2018-03-18 17:16:32,636 - DEBUG - [(131717719598912)-MainThread] CuraEngineBackend.CuraEngineBackend.slice [218]: Build plate 0 has no objects to be sliced, skipping
Segmentation fault (core dumped)
**Expected results**

Should load file and slice it
**Additional Information**

Link to STL file:
https://www.thingiverse.com/download:1779457

All the other STL files on that thingiverse page load and slice without a problem. This one will not load - Cura just segfaults.

ChrisTerBeke, 2018-03-18:
Your description let's me thing the file is not a valid STL. Have you tried pulling the file through something like Meshmixer and then loading it in Cura?

biscuitlad, 2018-03-18:
Meshmixer is a Windows program. I can't run it on Linux.

fieldOfView, 2018-03-18:
Then try Meshlab

biscuitlad, 2018-03-18:
Many thanks for the Meshlab suggestion. That did indeed fix it. But couldn't Cura reject the file in a more friendly manner, say without suddenly disappearing?

ChrisTerBeke, 2018-03-18:
That would be better indeed. I'll re-open this ticket so it can be addressed.

fieldOfView, 2018-03-18:
It would be my guess this is an issue in numpy-stl

Ghostkeeper, 2018-04-16:
All 3 models in the linked Thingi load fine for me using 3.3. No crashes. Does it still happen in 3.3?

If I recall correctly, we updated numpy-stl in 3.3 for a different issue (loading STLs that contain multiple meshes).

ChrisTerBeke, 2018-05-23:
Cannot reproduce on 3.3 anymore indeed. Closing.

closed 2018-05-23 by ChrisTerBeke