Ultimaker/Cura #3520: 3.2.1 Improve visibility for extruder selection please. Thank you!

xsk8rat, 2018-03-18:
**Application Version**
 3.2.1

**Platform**
 Windows 8
**Qt**
 Factory Version

**PyQt**
 Factory Version

**Display Driver**
 NVidia 7.8.840.0
Monitor = 3840x2160 (if that helps)

**Printer**
 UM3

**Steps to Reproduce**
 Use dual extruders and observe the GUI indications.

**Actual Results**
 Cura 3.2.1 has some important state indicators that are challenging to see and interpret. It would be very helpful if these were a more visually obvious. 

**Expected results**
 The selection of the extruder for a part should be clearly indicated on the screen.
For example: 
• When selecting which nozzle for a specific part, the difference between the selected nozzle and the non-selected nozzle is just a few pixels in size and a change in the line segments. 
• Selecting the nozzle (on the right of the screen) for the configuration of each nozzle is also very subtle. Just the addition of the cyan inlay on the button. 

**Additional Information**
 
Cura 3.2.1 has some important state indicators that are challenging to see and interpret. It would be very helpful if these were a more visually obvious. 

For example: 
• When selecting which nozzle for a specific part, the difference between the selected nozzle and the non-selected nozzle is just a few pixels in size and a change in the line segments. 
• Selecting the nozzle (on the right of the screen) for the configuration of each nozzle is also very subtle. Just the addition of the cyan inlay on the button. 

I have started prints with the wrong materials on more than one occasion - even though I am aware of my poor vision. 

It is possible to toggle the display using the "Layer View" option. But this is not the default setting for the "View" options, and there is a lag in this display as the object is re-sliced. 

Thank you for your consideration. 

Sincerely, 

Mark

ianpaschal, 2018-03-19:
@LukeChen-Ultimaker

Ghostkeeper, 2018-04-16:
The model is coloured darker if it is not the active nozzle. I think that is fine (but I know where to look).

The active extruder on the right-hand side of the screen is much more subtle and a very serious problem. Even our developers make this mistake sometimes. How to make it more clear though, I don't know. Maybe we could change the whole background colour of the sidebar based on the selected extruder?

xsk8rat, 2018-04-18:
I think that the concept of what you have implemented is in a acceptable direction.
But it might help if there was a way to make the change in color occupy _more_ of the screen space. For example:
[image]
It certainly doesn't need to be complicated.
Thank you for the note on the other folks having trouble!

fieldOfView, 2018-04-18:
Better yet, make them look like tabs:

[image]

#fullcircle

LukeChen-Ultimaker, 2018-04-18:
Looks like it's Cura 2.6?

Ghostkeeper, 2018-05-07:
Yeah eventually the choice was made to turn them into buttons. I think the reasoning behind it was that there were too many layers of tabs underneath each other. There was Prepare/Monitor, the extruders and then the recommended/custom modes. The buttons still behave like tabs though.

ianpaschal, 2018-05-07:
> The buttons still behave like tabs though.

Then they should also look like tabs IMO.