Ultimaker/Cura #3521: missing Holes

HgSiegenthaler, 2018-03-18:
Hi there
Cura can see the two holes but I cant print the unit with the 2 holes.
The unit is constructed with Rhino5.
Regards
Hg

[image]
[image]
[image]

ChrisTerBeke, 2018-03-18:
Can you show the mesh fixed category settings?

HgSiegenthaler, 2018-03-19:
This one?

Attached also the Cura project file.

 

Regards

Hg

ChrisTerBeke, 2018-03-19:
GitHub doesn't support sending attachments via email, please login to GitHub and upload the project file from there.

HgSiegenthaler, 2018-03-19:
[image]

How can I upload a Project?

ChrisTerBeke, 2018-03-19:
rename to .zip and then drag and drop in here.

HgSiegenthaler, 2018-03-19:
[Hole Test.curaproject.zip](https://github.com/Ultimaker/Cura/files/1824769/Hole.Test.curaproject.zip)

diegopradogesto, 2018-03-19:
Something looks wrong in the 3D model. Probably some normals are inverted. Very typical with Rhino. 
You can see a red area in the solid view:
[image]
Also using Meshmixer for instance you can see some errors in the model:
[image]

You need to fix it first.

closed 2018-03-19 by diegopradogesto