Ultimaker/Cura #3522: [3.2.1] Estimated time remaining counting up

nrgbistro, 2018-03-18:
**Application Version**
Cura 3.2.1

**Platform**
A Tarantula running Marlin software

**Qt**
---

**PyQt**
---

**Display Driver**

**Printer**
TEVO Tarantula

**Steps to Reproduce**
1. Send print to printer over USB

**Actual Results**
In the monitor screen, time remaining starts at 'Printing Time' and counts up

**Expected results**
In the monitor screen, time remaining should start at 'Printing Time' and count down

**Additional Information**
[image]

Print has been going for about 20 minutes; time remaining is higher than total print time. When print finishes, time remaining will be at about 8 hours, which does not make much sense...

nrgbistro, 2018-03-19:
After pausing and un-pausing for an unrelated reason, the timer seemed to fix itself. However I don't think this would have happened if I did not pause the print.

alekseisasin, 2018-03-23:
Hi @nrgbistro , I tested and I saw strange estimate time left counting, I created ticket for this bug.
Devs: CURA-5145

alekseisasin, 2018-06-29:
Hi @nrgbistro , the issue was fixed long time ago. You also can check it in version 3.4. Thank you.

closed 2018-06-29 by alekseisasin