Ultimaker/Cura #3523: Printer build plate settings

chputney, 2018-03-19:
Cura 3.2.1

Loading a model which is near the build plate size results in "unable to slice".  Here is an example.  I have an stl cube that is 200 X 200 X 100.  My actual build plate is 200 X 200 and the printer is "Custom FDM Printer".  If I set the printer settings on the actual build plate size of 200 X 200, when the model is loaded the result is "unable to slice".  In order to get the model to load, the build plate size must be increased to 225 X 225. (224 X 224 results in "unable to slice").  It seems that Cura has a "hidden" margin of 25mm.  Cura should respect the build plate size settings without this "hidden" margin or these hidden margin settings should be visible in the printer settings and adjustable.

A small annoyance is that when a model is unable to slice, it is positioned off the build area.  It would be nicer if the position (Move) was set to 0.

Vandrasc, 2018-03-19:
Hi @chputney, did you try to set the adhesion to None? The build plate volume is dependent on some settings (eg Avoid travel distance, build adhesion etc.)

chputney, 2018-03-19:
I have changed the build plate adhesion to none, and now the build plate needs to be set to 220 X 220 to fit a 200 X 200 shape without the "unable to slice" warning.

jackha, 2018-03-19:
There are certain disallowed areas at the sides and the corners (bed clips, switching bay, prime blob area) where the model cannot come. So, with a square model that fills the whole area will never fit. The prime blob can be turned off and you'll get a little more in the lower left corner.

chputney, 2018-03-19:
The Custom FDM printer should be able to define the build area and the unprintable margin.  On my custom printer case (and I suspect many others) there are no clips, switching bay etc.

jackha, 2018-03-20:
@chputney Ah sorry, I was talking about the UM3. The Custom FDM printer has no such disallowed areas. After turning off adhesion type, I can fit a 98x98x98mm cube in my build volume of 100x100x100mm.

chputney, 2018-03-20:
This is getting curious.  I get different results.  I will list the exact steps.
Using Cura 3.2.1, and restart Cura to get a fresh start.
Define Custon FDM with build area 100 X 100 X 100, Origin at center not checked, and set Xmin, Xmax, Ymin, Ymax to 0 (just to rule these setting out)
After defining printer, the camera view shows greyed out unprintable area at edges of about 9mm
The new printer definition was set to Brim, set this to none.
Now there are no greyed out areas on the camera view.
Load 84 X 84 block.  Result is "unable to slice"
Select the model and delete it.
Load 83 X 83 block. Result is that it loads successfully.
[blocks.zip](https://github.com/Ultimaker/Cura/files/1828753/blocks.zip)

And here's another interesting thing.  Once the 83 X 83 block is loaded, I can use the move to move it within the build area and show that the entire area is printable.

chputney, 2018-03-20:
Okay, I think I see the issue.  Loading a 98 X 98 block results in "unable to slice" but if the model is centered, it is now okay.  So the problem seems to be that when a model is loaded that is near the size of the build plate, it is bounced off the table and not centered.

BagelOrb, 2018-03-20:
I agree this is an annoyance and also that it is small. The fix should also be small though. "If number of items on build plate == 1 then center model when it's loaded"

Ghostkeeper, 2018-04-16:
There is also the "Avoid printed parts when travelling" that puts a small border around the edge.

SolidWorksMagi, 2019-03-22:
Most parts load and center on the build table reference in CURA 4.0, but some do not ???
[image]

Most parts load touching the build table reference in CURA 4.0, but some do not ???
[image]

Ghostkeeper, 2019-03-26:
The second problem is a different bug. Please report it in a separate issue. I'm unable to reproduce it but it looks like a bug.

Your first problem is the same as the original problem here. Actually, that's working as intended so I'll close this issue.

closed 2019-03-26 by Ghostkeeper

SolidWorksMagi, 2019-03-26:
Hi, 

 

Considering that CURA is the only GCode program causing the filled in spaces I think it's a bug in CURA.  My other GCode programs render the GCode correct for 3D printing.

 

 

 

==============================================================================================
C U L8r,  °|°  Walt Perko  °|°        "Kids ... teach them the good stuff, and they still learn the bad stuff on their own."

http://www.R2Pv1.com/  RoboGuts™ Intelligent content for 3D printing making S.T.E.A.M. education better, easier and more affordable  

 

Experiments to learn how to use various Electronic Components, Structured Computer Programming, Phonemes for Speech &Song in any language, and Art.

                         "The World Needs a New Economic Model"
==============================================================================================

BillyBotAvatar-2centsStamp-c

            Sent from the Cyber7

BagelOrb, 2019-03-26:
@SolidWorksMagi "filled in spaces"?! That doesn't seem to be related to this issue ticket.

If you still have an issue than please create a new issue and clearly describe the issue, clearly describe the steps to recreate the issue and preferably attach a project file.