Ultimaker/Cura #3525: Add Multi-build plate total estimate time and save all files

MiguelAngelLV, 2018-03-19:
I think that will be interesting add the total time of all build plates and the possibility of save all files as "project-name-1.gcode", "project-name-2.gcode"... Instead of save each plate and sum manually all estimate times.

diegopradogesto, 2018-03-19:
@jackha maybe has some ideas.

fieldOfView, 2018-03-19:
It would also be nice to be able to generate one big gcode file with a (configurable) gcode snippet in between each plate that moves away the head, cools down the hotends (to prevent material degradation), prints a message like "Please remove the prints", and pauses the printer before continuing with the next print. Sort of a poor-man's queueing system.

jackha, 2018-03-20:
@fieldOfView That's actually quite cool :-)
@diegopradogesto It should be fairly easy, for every developer ;-)

Ghostkeeper, 2018-07-04:
This idea was rejected by our project manager. We won't be able to spend time on implementing it soon.