Ultimaker/Cura #3528: Gaps/Voids between printed walls, doesn't happen on 3.1.0, no under extrusion nor not enough top layers.

ReplayMd, 2018-03-20:
**Application Version**
Cura 3.2.1
**Platform**
Windows 10
**Printer**
Creality cr-10 30x30x40

Last 2 weeks I was testing new filaments and colors. I am not the most expert 3d printer guy since I started using my cr-10 heavily the last 1-2 months. I thought it was a filament issue so I was testing with a lot of options.  While I am writing this sentence, I just checked that I had installed Cura 3.2.1 on 07 March, 13 days ago and pretty much everything starts making sense now. 

In the picture below you can see that I am having gaps between the walls & between the walls and the top layers. The specific prints are with 0.8 layers height and 8 top/bottom layers. The design is less than 30 cm in height/width. 

 Couple hours ago I installed again 3.1.0 and printed again the same design, same layer height and top bottom layers, same temp, same flow and most of the other setting stock. And there are no gaps like the latest version. 

[image]
[image]
[image]
_Although the picture if a bit out of focus, there are no gaps between walls like the previous photo but I can pretty much say that its an ideal top surface for me._ 

I am having the same issue of gaps/voids between walls, and between walls and the top surface with pretty much every print I got, with any layer height I tested(0.8,1.2, 1.6. 2.0)

**Any suggestion to fix in the latest version (3.2.1) or shall I just keep using the 3.1.0?**

smartavionics, 2018-03-20:
Could you please zip up the gcode for those two examples and attach it here? Thanks.

ReplayMD, 2018-03-20:
@smartavionics I have attached the two gcodes on the bottom my first post.  Thank you.

smartavionics, 2018-03-20:
Oops, yes, sorry, I didn't see that.

I have looked at the gcodes but can't see anything really obviously wrong with the 3.2.1 version. One difference is that you appear to be using coasting but did not in the 3.1 version.

Because you are using absolute extrusion mode it's difficult to compare the gcode files, could you please slice them again using relative extrusion mode (don't try to print them unless you know your printer can handle it) and attach a zip of those files. It may be easier to spot where the difference is. Thanks.

ReplayMD, 2018-03-20:
@smartavionics Yes, was testing coasting yesterday and the effects of it. Although its magnitude was only visible on the slicer and not the actual print, probably had to bump the numbers up a bit. 

I will slice them and also print them (I will try to match the settings as well in both versions) and post the gcode files and the pictures later today. Thank you.

ReplayMD, 2018-03-20:
@smartavionics I changed back a lot of my settings on the latest cura and I am having trouble replicating the gaps between the walls. Both versions seem to print pretty much the same now. Since the design takes from 7 mins to 15 depending the layer height, it's kinda small to notice if there are any gaps left anymore. Feel kinda stupid now if I consider that I sabotaged my self? (the beginning of the whole quest started from under extrusion of the new filament, caused by friction inside the bowden tube, even a 140% extrusion multiplier wouldn't help) 

I will report the later days on bigger prints to see if there are any gaps, or if it was actually my settings all this time. Thank you. :)

smartavionics, 2018-03-20:
Hi @ReplayMD , no problem. Why don't you close this issue for now. You can always reopen it again later if there is still something not right.

Regarding underextrusion, something that I only recently became aware of, even though it's pretty bloody obvious is that the faster you print, the more underextrusion you will get. Everybody probably knows that if you try to print too fast, the filament will strip but what I didn't realise is that the underextrusion also happens at lower speeds as well. I couldn't work out why I was getting underextrusion for some parts of my print but not other parts until google showed me this:

http://www.extrudable.me/2013/04/18/exploring-extrusion-variability-and-limits/

And now I have a solution because the printer controller I use (a Duet board) implements non-linear extrusion which compensates for the increasing underextrusion you get as the speed increases. It's really cool and made a small but visible difference to the print quality.

ReplayMD, 2018-03-20:
@smartavionics Thank you very much. I will check the article and take notice everything I learned these 2weeks!

closed 2018-03-20 by ReplayMD