Ultimaker/Cura #3530: InFill lines direction are locked to printer orientation.

Thisismydigitalself, 2018-03-20:
How could I have missed that!
InFill is printer oriented and not part oriented. I can understand the reasoning when printing on Cartesian printers but on my delta i wish to lock infill structure orientation to my printed part coordinates regardless of how i place it on bed. I know I can change that but wish to lock it to model - can I?

ozfunghi, 2018-03-20:
I know there is a hidden setting "infill line directions" but i don't exactly know how it works.

ozfunghi, 2018-03-20:
[image]

Thisismydigitalself, 2018-03-20:
This changes the orientation but does not lock it to model coordinates.

ozfunghi, 2018-03-20:
No, but it gives you  the option to change the orientation to the same directions as your model, as i understand it. So you could still get the result you want. Or not?

smartavionics, 2018-03-20:
>InFill is printer oriented and not part oriented.

Not true and if you want to move the infill within a part, use  "Infill X Offset" and "Infill Y Offset".

See how the infill is in the same position within each of these objects. The infill doesn't move within the object when the object is moved on the build plate.

[image]

Thisismydigitalself, 2018-03-20:
I am not moving my object around while keeping orientation - i rotate it.
And so far i did not find the way for the infill pattern to lock to the model.

smartavionics, 2018-03-20:
Oh, yeah, you're right, the infill angles are not relative to the model. Sorry, my mistake.

However, you can  still alter the infill angles and you can set the infill angles on a per model basis so it is possible to achieve what you want. It just means that you would have to alter the infill angles for an object after it has been rotated on the build plate.

ozfunghi, 2018-03-20:
That's what i was talking about. Could you give an example to how to use that feature? Do you just fill in a number for the angle?

Thisismydigitalself, 2018-03-20:
Thank guys. I did mentioned in my first post of this thread that i know how to change that but this by all means should not be the solution but the trigger to fix this. infill is part of the model - not the printer. I do understand that on Cartesian printers you might want the infill to follow X & Y axes but for Delta it should not matter. i find it hard to believe i missed that till now. i came across this behavior now when i had to rotate a big part which must be identical to a part i already printed but i need to rotate it around for better documenting the print progress with my camera.

smartavionics, 2018-03-20:
The Infill Line Directions setting just takes a list of angles surrounded with [ and ]. The angles are used in succession and the list wraps around so in my example shown below the infill angles are 45, 0, 135, 0, 45, 0, 135, 0 and so on. You can also set the angles of the top/bottom (aka skin) lines.

smartavionics, 2018-03-20:
[image]

smartavionics, 2018-03-20:
I have thought about making the infill angles relative to the object before now but, if my memory serves me right, it's not trivial because (and I may be wrong here) the slicer (CuraEngine) does not get passed the object rotation. Perhaps @Ghostkeeper  or @fieldOfView can let use know the truth of the matter.

ozfunghi, 2018-03-20:
If you want to change orientation, like what he wants and like i intended, you should just fill out one number. If you rotate your model 45°, and want the infill in the same direction as before, all you need to fill out is [45]. If you fill out more numbers, he will change the direction on the next layer, which is usually not what you want when you are looking for a rigid infill.

Thisismydigitalself, 2018-03-20:
Such threads tends to lose track pass this point :).
If i was asked how infill should work i would say Infill must be locked to either world or local coordinates. Cura should present a checkbox under infill category for the user to decide.

ozfunghi, 2018-03-20:
In this case i don't completely agree, because how they implemented it, you even have more freedom.  The description provided and the way smartavionics made it sound is just much too complex for what 99% of users would use this feature for, imho. "Set the angle in which direction you wish to rotate infill". Boom. Clear. For more advanced use, read the following phonebook.

Thisismydigitalself, 2018-03-20:
My only hope is for others to agree with me that infill is a property of the part being printed and we should be able to lock it and for project file to remember our decisions.

ozfunghi, 2018-03-20:
I would agree if the current function would not have existed. But i prefer this, because it can do what you want, and be customized.

Thisismydigitalself, 2018-03-20:
All I'm saying is that we should be able to **lock** the infill pattern to object coordinates. 
"There ought to be limits to freedom" :)

BagelOrb, 2018-03-20:
You propose to add yet another setting for making something more easily workable for something that's already achievable.

I dont think that many users want to optimize the infill direction before rotating the object. I think the added setting will be more confusing because there is yet another setting users have to understand.

Also the implementation is not trivial.

Thisismydigitalself, 2018-03-25:
> You propose to add yet another setting for making something more easily workable for something that's already achievable.

I will be more than happy to learn how to achieve what's already achievable.

BagelOrb, 2018-03-25:
Just edit the infill directions setting.

Thisismydigitalself, 2018-03-25:
As I was saying from the very beginning of this thread - I know I can do this but this is not how this should work cause this changes the infill grid/pattern relative to printer's built plate coordinates so if you re-orient your sliced object - infill pattern changes in relation to the object. I might have missed your solution for how to overcome this. Cura is a serious slicing software. it should not overcome such an important and basic feature. I admit that I am not familiar with any other slicing software since i am dedicated to Cura these last 18 months give or take so i don't know how this is handled in other slicers but won't you agree that infill pattern have a profound influence on part strength and characteristics and therefore should be managed accordingly? Again, I might be wrong and I missing the point so it's still an open discussion and by all means not me forcing my views on anyone. It sure feels like I am the only one who keeps opening old projects and re-print parts which I printed in the past. sometimes you need to re-arrange parts on built-plate. as things are right now - you should avoid that at any costs, that is if you want to preserve your project's settings.

smartavionics, 2018-03-26:
I agree that the infill directions should be relative to the part, just like the infill offsets are. As I mentioned above, I have considered this before and the only thing that stopped me doing it back then was the fact that I did not know how to obtain the part's rotation. Once that angle is known, i don't think it is much effort to implement. Also, I  would not bother adding another setting, I would just apply this to all parts.

Ghostkeeper, 2018-04-19:
> I have thought about making the infill angles relative to the object before now but, if my memory serves me right, it's not trivial because (and I may be wrong here) the slicer (CuraEngine) does not [...]

Indeed, CuraEngine does not know the rotation. It's complex and ambiguous to know the rotation after you've rotated around the Z axis 45 degrees, rotated around the X axis 175 degrees, then rotated around Z -15 degrees, then mirrored your object, then rotated around the Z axis 2.5 degrees once more, etc... You'd still have it change the locations where the infill lines connect if you combine a rotation around the Z axis with a rotation around the other two axes. Or a mirror. Or scaling. And it becomes more complex once you group two objects and then rotate the group. We only multiply the model coordinates in the front-end and then we apply all of these changes at once with a 3D transformation matrix.

So I'm going to say that this feature request is impossible or at least unintuitive.

closed 2018-04-19 by Ghostkeeper