Ultimaker/Cura #3531: Feature request: maximize brim/raft

ozfunghi, 2018-03-20:
An idea for items that would not leave any space for a traditional raft or brim. 

The images will make things more clear than me trying to explain it. When you are reaching the limits of your bed size, that means a brim or raft is no longer an option, because it would exceed the bed dimensions. But often there is still plenty of room on the bed for a meaningful brim or raft. In many cases this could still be very helpful.

[image]
[image]

White plane =  bed
black shape = 3D model
grey shape = raft/brim

BagelOrb, 2018-03-20:
Currently the brim.consists of concentric lines around the model, but thats not possible with your idea, so I wouldnt call this 'brim' anymore I guess.

This also means that it will be harder to remove and it will leave more blemishes.

I'm not sure the Cura team will spend time on this. For the moment you could just model a thin piece with the height of the jnital layer and print that as brim.

ozfunghi, 2018-03-21:
Also, i'm open for creative suggestions. Maybe a feature that adds many small "blobs" close to the model (see image):
[image]

Ghostkeeper, 2018-04-16:
We're investigating the possibility of removing brims in inside holes even if those inside holes have something else in them. This requires being able to cut off brim lines where they don't fit. This is also a requirement for your solution, so maybe once we have that, your solution comes into the realm of possibilities.

BagelOrb, 2018-04-16:
It doesn't necessarily involve cutting off brim lines. You could also
generate an area which to fill concentrically by performing an offset from
all outer contours and doing an area difference with the fill first layer...

GregValiant, 2024-11-14:
This feature is now included in Cura 5.8.0 and up.  I'll go ahead and close this.
[image]

closed 2024-11-14 by GregValiant