Ultimaker/Cura #3532: [3.2.1] Raft Air Gap is behaving incorrectly.

megablue, 2018-03-21:
**Steps to Reproduce**
Pictures explain it better https://imgur.com/a/PoHIR

Step A
1. Change the raft air gap to 0
2. The first layer of the model begins at 0.8mm

Step B
1. Change the raft air gap to 0.1mm
2. The first layer of the model still begins at 0.8mm 

**Actual Results**
Step A
1. Change the raft air gap to 0
2. The first layer of the model begins at 0.8mm

Step B
1. Change the raft air gap to 0.1mm
2. The first layer of the model still begins at 0.8mm 

**Expected results**
Step A
1. Change the raft air gap to 0
2. The first layer of the model begins at 0.8mm

Step B
1. Change the raft air gap to 0.1mm
2. The first layer of the model should begin at 0.8 + 0.1mm 

**Additional Information**
I had the initial Z overlay set to 0 as well

BagelOrb, 2018-03-21:
You're probably looking at support. Support doesn't apply the airgap.

BagelOrb, 2018-03-21:
[image]

closed 2018-03-22 by megablue