Ultimaker/Cura #3533: Custom Printer Filament Length Usage Not Correct

chputney, 2018-03-21:
Cura 3.2.1 on Windows7

The material length estimate for a custom FDM printer shown after a model is sliced by Cura is incorrect.

To demonstrate: 
Create a Custom FDM printer and change the extruder size to 1.75mm

Set the print mode to recommended and the layer height to 0.1mm and material to PLA.
Load the model helix.stl
After slicing, the length of filament used is reported by the UI to be 0.94m
Open the resulting helix.gcode file and it shows that the filament length used is 2.49m

Apparently, even though the filament diameter was changed from 2.85mm to 1.75 in the printer definitions , the UI is still using the 2.85mm because:

(0.94 * 2.85 * 2.85) / (1.75 * 1.75) = 2.49

So it looks like the UI view is somehow getting the wrong infomation about the extruder size.

Printing this model with slightly different settings resulted in a filament usage of about 2 meters, so the estimate of 0.94 meters is definitely wrong.

[helix.zip](https://github.com/Ultimaker/Cura/files/1833650/helix.zip)

alekseisasin, 2018-03-23:
So far I know, the there was some ticked which referred to the wrong material length. Using development version I see the filament length 2.49. According to provided calculations:
(2,49 * 1.75 * 1.75) / (1.75 * 1.75) = 2.49

chputney, 2018-03-23:
I'm not sure that I understand.  The UI seems to get the correct material volume from the slicing engine, and then calculates the length using the wrong filament diameter of 2.85.  It seems to me that although the user has changed the filament diameter to 1.75mm, the calculation uses filament diameter 2.85.

When you refer to "development version", do you mean that this is the version that will be released in the future, so this bug is fixed ?

Ilya-VK, 2018-03-25:
Confirmed: Cura 3.2.1, Windows 10 with all current updates.

Bug description: filament length displayed in UI near Save file... button is incorrect.

Details: filament weight displayed is correct or near correct. Taking in account that I'm not sure in filament density exact value, it's in +/-10% tolerance and OK. But required filament length is diplayed wrong, almost 3 times less than actually needed.

For example, for 1.75 mm PET-G with density of 1240 kg/m3 for a model of 12 gramms only 1.47 m of filament is displayed. Verification calculation:

1) mass of 1 meter of filament:
1240 [kg*m^-3] * 1 [m] * (0.00175 [m]/2)^2 * pi = 0,002982... [kg]
or
approx. 2.982 [g/m]

2) required filament length for 12 g print:
12 [g] / (2.982 [g/m]) = 4.023 [m]

3) error coefficient: 4.023/1.47 = 2.737

I failed two prints because of this bug loading not enough filament, not expencive, but was disappointed anyway.

fieldOfView, 2018-03-25:
This is fixed for Cura 3.3 in #3285

closed 2018-04-17 by Ghostkeeper