Ultimaker/Cura #3535: [3.2.1] - Multi component Print names

UltimateSports, 2018-03-22:
I think it could be helpful for many of us if when we have multiple parts on the print bed that are subtly different if the part name would pup up on the geometry arrows or in the bottom left as a list that highlights the name of the part selected as it is done on Modeling Software assembly modes. 

need to be able to check that I have the latest iteration, or check the multiples of components etc especially if its a long print and come to it afterward to have printed many multiples of the wrong part...

alekseisasin, 2018-03-23:
To see names of the loaded models you might try experimental feature called: "Use multiple build plate functionality", it might help get overview of all loaded models on the build plate. 
I see you expect sort of additional data on the screen, maybe you can make a draft to better understand what it can be.

ianpaschal, 2018-03-23:
I would try @alekseisasin's proposal for now but I agree that this could be a very nice feature to have, especially on printers with larger bed sizes where you want to print lots of small parts.

alekseisasin, 2018-03-25:
It is best to start from draft, Can you add some comments: @LukeChen-Ultimaker ?

alekseisasin, 2018-06-29:
@LukeChen-Ultimaker , @AnneUltimaker , ok if we agree to add this option. 
Simple scenario. Every time when I select the model the name will be updated. If it will be in right corner then it might mislead users to distinguish the file name and model name. I think we might display the model name in a different way. Maybe cursive, or with lower opacity. ?

fieldOfView, 2020-03-23:
There is now an object list, which highlights the currently selected object(s)

closed 2020-03-24 by Ghostkeeper