Ultimaker/Cura #3536: [3.2.1] No LOAD PROJECT option in menu - Linux

andyspoo, 2018-03-22:
**Application Version**
3.2.1

**Platform**
Linux Mint 18.3 Sylvia

**Qt**
5.5.1

**PyQt**
5.5.1

**Steps to Reproduce**
Every time I go to FILE, the LOAD PROJECT option is missing. Save project is their. Cannot load project through standard 'Open Files', but that is as expected.

**Actual Results**
Can Save, but not reload a Project.

**Expected results**
Should be able to load a Project, if it has the option to save a project.

**Additional Information**
[image]

BagelOrb, 2018-03-23:
I find the interface a bit confusing Inthis respect as well.

You should be able to open project by just clicking [Open...] and selecting the project file.

If that opens the model without the settings, you should change that behaviour in the preferences.

BagelOrb, 2018-03-23:
I still think this is a valid feature request for having a load project, which always loads the .curaproject.3mf file as a project.

alekseisasin, 2018-03-23:
I my view point "Open File(s)" allows open a project or *.stl file, which should be fine. If we would have "Open project", how will it improve your working process? For example, SolidWorks has a button in menu "Open", which allows open one model or assembly and I think it is right.

alekseisasin, 2018-03-26:
No. 
By default in Cura we start such a assembly project. Which can be translated/saved as project or model. 
By fact, It is not hard to implement, but does it make UI  simpler for the users, from my prospective _no_. (For now)

alekseisasin, 2018-06-29:
@LukeChen-Ultimaker , @AnneUltimaker ,  Could please share your opinion.

Ghostkeeper, 2018-08-10:
On Master we don't have a "Save Project" entry any more now. We have "Save" which saves the project and "Export" which saves the scene as model files or g-code. There is also "Load" which loads all files: Models, g-code or projects.

Isn't that a bit more intuitive?

LukeChen-Ultimaker, 2018-08-10:
The original wished function: "load project" is achieved through the option: "Open files" with current Cura 3.4 version. From UX point of view, the issue is resolved for now.

BagelOrb, 2018-08-10:
A normal interface has [Import] and [Open] as two separate options, rather than putting them both under [Open]. 

Is there a reason we put them under the same menu item?

closed 2020-11-03 by nallath