Ultimaker/Cura #3537: master branch - layer view slider for selecting line stuck on 100%

smartavionics, 2018-03-22:
**Application Version**

master branch

**Platform**

Linux

**Qt**

5.8.0

**PyQt**

5.8.2

**Display Driver**

2018-03-22 11:27:02,763 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [92]: Initialized OpenGL subsystems.
2018-03-22 11:27:02,763 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [93]: OpenGL Version:  4.5 (Core Profile) Mesa 17.2.4
2018-03-22 11:27:02,763 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [94]: OpenGL Vendor:   Intel Open Source Technology Center
2018-03-22 11:27:02,764 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [95]: OpenGL Renderer: Mesa DRI Intel(R) HD Graphics 520 (Skylake GT2) 

**Printer**

Kossel Mini

**Steps to Reproduce**

Load model, slice, select layer view, try to move line selection slider

**Actual Results**

Slider bug tries to move but then jumps back to 100% immediately.

**Expected results**

Slider bug moves like it used to and you can select the line currently being drawn.

**Additional Information**

Not sure when this change occurred but it must be pretty recent as I use the slider a lot.

smartavionics, 2018-03-22:
Found it, https://github.com/Ultimaker/Cura/commit/da00bdf9c325e9e124ea1c20f351470bf01e3b4e is the culprit.

BagelOrb, 2018-03-23:
@ghostkeeper should be an easy fix then.

smartavionics, 2018-03-23:
You would have thought so, whaddyathink @LipuFei  ?

LipuFei, 2018-03-23:
Perhaps we need to connect to both signals.

smartavionics, 2018-03-23:
Perhaps. I have no opinion on that, it's all black magic in the front end!

smartavionics, 2018-03-23:
Thanks @ianpaschal, closing...

smartavionics, 2018-03-23:
Err, I think that was @alekseisasin who comitted the fix, thanks!

closed 2018-03-23 by smartavionics

alekseisasin, 2018-03-25:
I am not sure about connecting both signals, the onSceneChange is triggered constantly ( after each mouse move), This _self._controller.getScene().sceneChanged.connect(self._onSceneChanged)_ , is correct it was before.
If you apply this:
_self._controller.getScene().getRoot().childrenChanged.connect(self._onSceneChanged)_
Then simulation is broken, try to click play button.