Ultimaker/Cura #3539: How to setup Diamond 3 color extruder?

mubarak111nsu, 2018-03-23:
Dear respected sir, I am using  Diamond 3 color extruder where there is 3 extruder and a 3 in and 1 out nozzle. 
Here in Cura, I see 2 extruder> Extruder 1 and Extruder 2
Now to set up my extruder in Cura , What I need to do?

[image]

ianpaschal, 2018-03-23:
Cura doesn't currently support 3-to-1 extruders.

But you can fake it pretty well by adding a custom FDM machine (instead of a UM3) and then setting the number of extruders to 3.

The caveat is that when one extruder is printing, the others will be set to the standby temperature which in this case, is the same nozzle, causing it to cool down too much to print. So you'll need to set the standby temperature for all nozzles to be same as the printing temperature. This of course means you can only use the same material across the board but this isn't such an issue if your goal is to just print different colors of PLA.

mubarak111nsu, 2018-03-24:
Dear sir, I have solved the problem. Thanks a lot

closed 2018-03-25 by ianpaschal

madmax2, 2018-07-08:
@mubarak111nsu how?