Ultimaker/Cura #3546: Dropdown for "Save to file" or "Print via USB" is not shown on full screen (maximized window)

shivan, 2018-03-23:
**Application Version**
3.2.1

**Platform**
Windows 10 Fall Creators Update

**Qt**
using ultimaker version

**PyQt**
using ultimaker version

**Display Driver**
nVidia GeForce GTX 1060 - driver version 23.21.13.9077
Intel HD Graphics 530 - driver version 22.20.16.4749

**Printer**
Creality CR-10

**Steps to Reproduce**
Set "manual prepare" in the settings
plug printer to USB
click "Prepare"
Then there is a dropdown which should show 
"Save to file" and "Print via USB"

**Actual Results**
But if the window is full screen then it does not appear. Only a black line appears:
[image]

When window is not full screen it appears:
[image]

**Expected results**
The dropdown should drop "up" -- should be at least visible ....

**Additional Information**
I'm using a dual monitor setting. This was tried on the main monitor. (1920x1080) Second is 1600x1200.

alekseisasin, 2018-03-26:
This is such a weird a bug, which might depend on any windows update or QT (the framework which us used by Cura for the UI) update. I use Ultimaker 3, Windows 10, but my graphical card not so strong as your and I still can see all items in the drop down.
Coming days we will release Cura 3.3 Beta version, you might try it. If it will not work properly for you then we will keep in mind your issue.

shivan, 2018-03-28:
Tried 3.3.0-BETA. Still the same.

Ghostkeeper, 2018-04-25:
I came across a Qt bug report for this: https://bugreports.qt.io/browse/QTBUG-42267

But they claim it was fixed in version 5.6. We're currently at 5.8.

shivan, 2018-04-26:
Tried 3.3.1 - still the same...

[image]

Ghostkeeper, 2018-05-25:
Only not on our testing computers :disappointed:

Maybe it has something to do with Windows themes? Do you happen to have a dark theme?

shivan, 2018-06-04:
Yes, I have the Dark Theme on Windows 10 Spring Creators Update.

Device is a MSI Notebook with graphics card:
* nVidia GeForce GTX 1060 (Driver: 07.05.2018)
* Intel HD Graphics 530 (Driver: 24.07.2017)

I have a dual screen setup. Notebook and on the right an additional 1600x1200 screen.

ilarischeinin, 2018-09-26:
I have this issue on macOS too (High Sierra 10.13.6 and Cura 3.4).

Ghostkeeper, 2018-11-22:
This issue will be fixed with the next release. We didn't actually find what caused this bug or fix this bug ourselves, but that drop-down is no longer an actual system drop-down in the next release. We just render it ourselves.

closed 2018-11-22 by Ghostkeeper

Ghostkeeper, 2018-11-22:
See https://github.com/Ultimaker/Cura/pull/4715