Ultimaker/Cura #3548: How to setup Marlin firmware with RUMBA broad and Diamond 3 in and 1 out extruder?

mubarak111nsu, 2018-03-24:
Dear respected friends, I am new in 3D printing. 
For last few months, I am working on 3D printing. In my 3D printer, I am using RUMBA bord and Diamond 3 in and 1 out extruder. 
I am also using Marlin firmware and Ultimaker Cura. 
MY Printer dimension is X-axis 1000mm, Y-axis 700mm and Z-axis 600mm 
But I am facing some problem to setup Marlin firmware. Please sir, I am begging for help.

Please sir, Help me. I will be grateful to you
[image]

jackha, 2018-03-26:
Hi @mubarak111nsu 
I'm sorry to hear you have problems with your marlin firmware, but here we're only doing Cura related issues. Please refer to the manufacturer (or designer) of your printer on how to set up your Marlin. Good luck!

closed 2018-03-26 by jackha

Ghostkeeper, 2018-04-19:
Try this place: https://github.com/MarlinFirmware/Marlin