Ultimaker/Cura #3549: printer setting bug

xia01eng, 2018-03-24:
[image]
As shown in this picture, every time you close the cura application the nozzle size configuration just cannot be stored. Thus every time I restart cura, I have to change that setting. It's really ... . Hope to fix that bug.

fieldOfView, 2018-03-24:
A screenshot shows us nothing in this case. Please locate and share your logs (cura.log in the configuration folder)

xia01eng, 2018-03-25:
Sorry, where to find configuration folder. OSX

xia01eng, 2018-03-25:
@fieldOfView

xia01eng, 2018-03-25:
Unfortunately, it did not display any information about this issue in the log. But just like I described before, this problem did exist.

Thisismydigitalself, 2018-03-25:
> this problem did exist. 

> 
Still exists in 3.2.1.
You mean fixed in 3.3? any ETA for 3.3?

ianpaschal, 2018-03-26:
Duplicate of #3511.

This is fixed and will be in the 3.3 beta released this week.

closed 2018-03-26 by ianpaschal