Ultimaker/Cura #3550: Improvement: Prime tower

mtmello2003, 2018-03-24:
Hello,

There is the condition that layer after layer, wont be exxtruder change but CURA still filling the prime tower with the same extruder (filament color) using the "prime tower minimum volume" (normally set to full - solid tower).

In that case, the better way is cura use a new parameter like: "prime tower infill %"

Will save a lot of filament and time because if do not have extruder change, do not need to fullfill the tower using the minimum volume...

thanks

Ghostkeeper, 2018-04-18:
It's probably fine to use the normal infill density of the appropriate extruder for this.

sjoerdtimmer, 2018-06-03:
I would love this too as my 2-in-1-out setup requires a large prime tower it would soothe the pain a little if the prime tower layers that don't actually change color would be printed like infill.
@Ghostkeeper, are you working on this?
If not I will try to implement this myself.

ianpaschal, 2018-07-02:
For reference: JIRA CURA-5248

Ghostkeeper, 2018-07-04:
I'm currently working on a new strategy for the prime tower that doesn't alternate the layers any more but prints one of the extruders inside the other, and causes the outer extruder to always print. This sort of invalidates this issue since the prime tower can be made much thinner because the adhesion is better.

I think it will be unnecessary.

For a demo you can try out the feature_prime_tower_shells branch on CuraEngine.

closed 2018-07-04 by Ghostkeeper

Ghostkeeper, 2018-08-31:
It is no longer necessary because the prime tower circles are now printed next to each other, so they can have different thicknesses. They no longer need to have the same thickness because they are not stacked on top of each other. So instead of adjusting the infill rate, you can simply adjust the thickness.