Ultimaker/Cura #3551: [3.2.1] Adpative layer + prime tower

mtmello2003, 2018-03-24:
Win10 x64

When I set "enable prime tower" plus "adpative layer", no layers are showing.

[image]

alekseisasin, 2018-03-26:
Hi @mtmello2003 , I did same steps and Cura it works fine. I see you are using Compatibility mode, maybe this is the case. Then the problem is related to your hardware.
I can test again if you will upload your project here.

mtmello2003, 2018-03-26:
Hi,

The stl file is from https://www.thingiverse.com/make:467481

Thank you!

alekseisasin, 2018-03-26:
@mtmello2003 , can you save model as project, then I will ensure that I do not miss any setting. To do this please go the menu bar and in "File" section choose "Save Project...".  This project you can archive and upload here.

diegopradogesto, 2018-03-26:
I don't get the crash, so maybe it's already fixed. Which version of Cura are you using? Next time, please also fill the form that is included when you create the issue, it contains very useful information for us to trace the error.
As @alekseisasin says, you project is very helpful also in this case.

mtmello2003, 2018-03-27:
This issue is completely vanish with cura 3.3 B.

Please close the call

Thank you

[image]

diegopradogesto, 2018-03-28:
Thanks for checking.

closed 2018-03-28 by diegopradogesto