Ultimaker/Cura #3552: Crashing cura 3.2.1 with setting Printing Temperature property to zero.

amidarius, 2018-03-24:
**Application Version**
Cura 3.2.1

**Platform**
Windows 8 Pro

**Steps to Reproduce**
If Default printing temperature is set to some value, say 215 and then set Printing Temperature to 0 to follow the default value application just crash.

**Actual Results**
  Problem Event Name:	APPCRASH
  Application Name:	CuraEngine.exe
  Application Version:	0.0.0.0
  Application Timestamp:	5a840e96
  Fault Module Name:	CuraEngine.exe
  Fault Module Version:	0.0.0.0
  Fault Module Timestamp:	5a840e96
  Exception Code:	40000015
  Exception Offset:	000000000020e212
  OS Version:	6.2.9200.2.0.0.256.48
  Locale ID:	9242
  Additional Information 1:	a91c
  Additional Information 2:	a91cefa0372c43ceda6c55f0cdec39f5
  Additional Information 3:	1084
  Additional Information 4:	1084e84f8d0b5a729749041e8725ee9e

**Expected results**
To follow default temperature.

ianpaschal, 2018-03-26:
Where are you setting default temperature? In the material profile? Also a crash is not expected but I also would not expect that if you override the default temperature with 0 that the default is used.

ianpaschal, 2018-03-27:
Duplicate of #3562; fixed.

closed 2018-03-27 by ianpaschal