Ultimaker/Cura #3553: Improvement: Different prime tower volume for each extruder

mtmello2003, 2018-03-24:
Win10 x64

Different type of filament colors need more or less volume to prime. For example: when I use white in E1 and Orange in E2, White need the double of volume to prime. Orange one need only  the half of parameter "Prime tower minimum volume" to be completely cleaned.

If we have the option to define different values for each extruder, could be perfect!

Thanks

diegopradogesto, 2018-03-26:
The "Prime Tower Minimum Volume" is already settable per extruder. However the engine doesn't use that value but the "Prime Tower Thickness" that is calculated using the former. The thickness is not settable per extruder and CuraEngine uses the maximum value of all the extruders.
I think the behavior is correct, using the maximum, otherwise when priming the white on top of the orange will be not enough material to support the tower and it will fall, right?
I don't see your point. Why not just setting the amount you need for the white material?

mtmello2003, 2018-03-26:
Let me try to explain giving an example:

If I set a Prime tower to 200x200mm and set to complete fill this area, the config will be applied for both extruders (both filament colors). 

In a case of using white + black filament: I do not need to fill entire 200x200mm using black. Black will be ok just with 30 or 40% of this area but when I exchange from black to white, I need to fill entire 200x200mm to clean it completely.

That´s the reason that if I have an option to define for the prime tower different values to fill it, I can save filament, time and get the same result at all.

It is a global rule: bright colors need more area to prime the nozzle when came from a dark color.

Actually, I need to set the Prime Tower thinking in the bright color, in despite that dark color do not need all that big area to do the same.

diegopradogesto, 2018-03-26:
Probably you meant that the position of the prime tower is x = 200, y = 200, not the size. Otherwise this is a huge prime tower.
Regarding your concern, I think I understand well what you want, but my question is why not to use the maximum of them (as it is now)? There is not a lot of waste of material or time and the prime tower will be well supported.

mtmello2003, 2018-03-26:
Sorry: not 200x200 but 20x20mm prime tower size.

In a tower with 10 layers could not make a big difference in waste of time and filament but if it increase to 100 or 200 layers, if count a really big waste of both.

The tower do not need to be empty but only use infill % (define by the user in a parameter) plus 1 shell or 2 shells.

By the way, thank you to spend your time answer this call and considering this subject!

BagelOrb, 2018-03-30:
Ok the prime tower could just be filled with infill with a given percentage. That way the strength of the tower is still guaranteed.

Ghostkeeper, 2018-04-26:
Duplicate of https://github.com/Ultimaker/Cura/issues/3550

closed 2018-04-26 by Ghostkeeper