Ultimaker/Cura #3554: Choix du matériaux

Mmanga78, 2018-03-24:
**Application Version**
 3.2.1

**Platform**

Windows
**Qt**

**PyQt**

**Display Driver**

**Printer**

Annycubic-i3 Méga
**Steps to Reproduce**

**Actual Results**

Je ne peux choisir pas mon matériaux et du-coup mon prix ne s'affiche pas en dessous du temps d'impression.
**Expected results**

Je suis sensé choisir mon matériaux parmi ceux proposer et donc  aussi changer mes paramètre en fonction du matériaux, le prix de l'impression est aussi sensé s'afficher en dessous du temps d'impression.
[image]

**Additional Information**

fieldOfView, 2018-03-24:
Duplicate of https://github.com/Ultimaker/Cura/issues/3266. See the explanation there.
Please note that the language here is english. The Ultimaker forums have a french section.

ianpaschal, 2018-03-26:
Thanks @fieldOfView 

@Mmanga78: Nous faisons de notre mieux pour prendre en charge le matériel de tierce partie, mais en fin de compte vous êtes à la merci de la tierce partie pour faire des profils appropriés.

closed 2018-03-26 by ianpaschal