Ultimaker/Cura #3555: [3.2.1] CuraEngine crashes when slicing with a raft and Number of slower layers = 0

MajorSlug, 2018-03-24:
**Application Version**
3.2.1

**Platform**
Windows 7

**Printer**
any

**Steps to Reproduce**
- load a STL file
- choose any profile (eg. low quality - 0.15mm)
- set Number of slower layers to 0
- choose "raft" Build Plate Adhesion Type
- prepare

**Actual Results**
CuraEngine crash message pops up and no slicing is done.

**Expected results**
Having the mesh sliced laying on a raft.

**Additional Information**
[image]

smartavionics, 2018-03-24:
Either the bug has been fixed or it's more subtle than just those two settings because I can slice using that combo with the current dev version of Cura without it crashing...

[image]

smartavionics, 2018-03-24:
Interesting that the displayed min feed rate is complete bollocks.

[image]

MajorSlug, 2018-03-24:
My machine is a "Custom FDM printer". I updated the issue.

smartavionics, 2018-03-25:
I do, however, see a couple of these messages which could be relevant:

2018-03-25 07:43:45,940 - WARNING - [MainThread] cura.PrintInformation._updateTotalPrintTimePerFeature [190]: Received NaN for print duration message

smartavionics, 2018-03-25:
Possibly fixed by https://github.com/Ultimaker/CuraEngine/pull/730.

MajorSlug, 2018-03-25:
I'm pretty sure it is not fixed by Ultimaker/CuraEngine#730.
The change being:
[image]
I don't see how layer_nr could be < 0.
Anyway, didn't you mean
`if (layer_nr >= 0 && layer_nr < initial_speedup_layer_count)`

smartavionics, 2018-03-25:
I think you will find that layer_nr will be negative while the raft is being printed. So I am stopping that function being called for the raft layers.

diegopradogesto, 2018-03-26:
I don't see my CuraEngine crashing with those settings.
However what I see is that all the feedrates (wall, infill, first layers, travel moves, ...) are the same. That's the reason why the model is all orange and why the min_feedrate is the float's minimum value. I'm more worried about that.
I will investigate a bit more.

smartavionics, 2018-03-26:
Hi @diegopradogesto , please check my engine PR that I believe fixes this issue.

diegopradogesto, 2018-03-26:
Yes, I will do that.

diegopradogesto, 2018-03-26:
The only concern with the PR is that the two first layers of the Raft will print slowly, not taking into account the "Number of slower layers"

diegopradogesto, 2018-03-26:
Yes, I will do it. I was just thinking aloud :-)

diegopradogesto, 2018-03-26:
This bug is fixed in the next release. Thanks @MajorSlug for the report and @smartavionics for the quick fix.

closed 2018-03-26 by diegopradogesto