Ultimaker/Cura #3556: Good afternoon. I'm just starting to explore the 3d world and want to ask a couple of questions about the Car Engine. Is it possible to get the model data (the file) when you load the model (for example, the weight of the model, the manufacturing model).

Creativ671, 2018-03-25:
Platform
Windows 10
Good afternoon. I'm just starting to explore the 3d world and want to ask a couple of questions about the Car Engine. Is it possible to get the model data (the file) when you load the model (for example, the weight of the model, the manufacturing model).

ianpaschal, 2018-03-26:
I don't totally understand what you are asking.

After you slice a model with Cura, and CuraEngine returns the gcode (machine paths), it also shows how much filament will be used by length, and by weight (or more specifically, by mass).

Creativ671, 2018-03-26:
I want to make an stl file calculator for the website using the Cura slicer. But I don't know how to do it.  (Sorry for my English)

ianpaschal, 2018-03-26:
Ah okay.

It's possible, but I'm not sure I can really explain how to arrange all your code and what to build via a GitHub comment.

I suggest looking [here](https://github.com/Ultimaker/Cura/tree/master/plugins/CuraEngineBackend). This is the Cura backend which is responsible for sending the meshes to CuraEngine and then listening for the machine code (gcode) response.

Hopefully that gets you started!

closed 2018-03-26 by ianpaschal

Ghostkeeper, 2018-04-18:
The print time and material usage is normally put in the g-code header.