Ultimaker/Cura #3557: Algorithm for 3D Supporters

AndreAhmed, 2018-03-25:
Would someone refer to me some algorithms for supporter generations ? a paper or anything related that is used in Cura or some information ?

ianpaschal, 2018-03-26:
I don't think there's a single paper upon which the functionality in Cura is based. I think you're better off reading the code in Cura Engine or simply researching 3D printing support.

That being said, the person who I would be the first to ask is @BagelOrb, as he's actively involved in PhD research in 3D printing.

closed 2018-03-26 by ianpaschal

BagelOrb, 2018-03-26:
I wrote the support generation algorithms, but it wasn't based on any literature and I didn't publish my work other than open source.

BagelOrb, 2018-03-26:
Here's the documentation:
https://ultimaker.com/en/resources/20422-cura-support-settings

AndreAhmed, 2018-03-26:
@BagelOrb Would you explain the algorithm ? 
My idea is to ray cast from a grid and see if it hits, then generate a support, but how would I determine the angle of the tip ? there are many questions.
I would be helpful to outline the algorithm.

BagelOrb, 2018-03-26:
The algorithm works on the polygon outlines of layers. Using raycasting
inevitably misses underhang points surrounded by steep surfaces.

Imagine a steep upside down cone. All of its faces would satisfy the
overhang angle, but still it would be printed in thin air.

The raycasting approach is simply not robust enough.

I'm not going to explain the full algorithm. That takes too much time.

The basic idea is to do an offset of the outlines below and subtract that
area from the layer above to get an idea of where there are overhangs.

AndreAhmed, 2018-03-26:
There is no solution for the steep upside down cone ? I have only the solution for the ray casting.

BagelOrb, 2018-03-26:
The ray casting algorithm has no solution for it. It's a fundamental
limitation of that approach.