Ultimaker/Cura #3558: Feature Request: Initial Layer Flow

zBeeble42, 2018-03-25:
**Application Version**
All
**Platform**
All0
**Qt**
N/A
**PyQt**
N/A
**Display Driver**
N/A (but it's nVidia 3.56 or there abouts)
**Printer**
I built it myself.  Roughly, it's a Mendel with Duet WiFi electronics
**Steps to Reproduce**
Stare wistfully at the interface and remark on it's absence.
**Actual Results**
I'm somewhat disappointed.
**Expected results**
OK.  Meat of the issue.  Initial Layer flow.  In my particular combination of circumstances, I have trouble with adheasion of small parts and small details of the first layer if I set the  flow at 95%.  However, with the flow at 100% (which fixes this) I get the occaisional goober in higher layers.  Goobers go away with the flow at 95%.

So far, this means making prints multiple times to get a sweet spot, but I'm convinced that an initial layer flow of 100 and a flow of 95 for the rest of the print will fix this.

**Additional Information**
That's pretty much it: I've said my peace.

smartavionics, 2018-03-25:
[image]

zBeeble42, 2018-03-25:
I'm confused now.  I've gone over the settings dialogue looking for the spot to enable this, and I cannot find it.

zBeeble42, 2018-03-25:
so ... here's my settings dialogue ... filtered to "flow":

[image]

smartavionics, 2018-03-25:
Hmm, well I don't know why it isn't visible for you. What version of Cura are you using? I think it has only been available this year.

zBeeble42, 2018-03-25:
It says 3.2.1 when it starts up.  Is this a Beta or somesuch feature?

smartavionics, 2018-03-26:
Looking at the commits, I can see that 3.2.1 was released after this setting was added so I don't know why it isn't showing up for you. I only work on the slicer back end, the front end UI is a still very much a mystery to me.

smartavionics, 2018-03-26:
Sorry, I was wrong, this setting was merged into the Cura main branch after 3.2.1. It will be in the next release which should be coming along as a beta pretty soon now.

ianpaschal, 2018-03-26:
I'm also not sure what's going on but I:

- Do not see the setting in my installation of 3.2.1
- Do see the setting when running from source

So I have a feeling this has something to do with the setting visibility preset functionality that we've been tweaking.

I'm closing this for now, because I'm fairly sure it will be fixed when running the beta released this week. @zBeeble42, if you're still having the issue after running 3.3 beta, we can reopen, but I'm pretty sure it will be fixed.

closed 2018-03-26 by ianpaschal

diegopradogesto, 2018-03-26:
It was a PR #2619 merged on January 24th, but it is not in the 3.2 branch. This setting will be ready in 3.3. We are releasing a beta soon.