Ultimaker/Cura #3561: UM Cura crashed and all settings were reset to default

Emergency1999, 2018-03-26:
**Application Version** 
Cura 3.2.1

**Platform**
Win 10

**Display Driver**
NIVIDIA GeForce GTX 1060 + 960

**Printer**
(UM3)

**Steps to Reproduce**
I don't know, crash the computer

**Actual Results**
All Cura settings were reset to default

**Expected results**
settings should NOT reset

**Additional Information**
Is there a way to get my settings back?

Thanks

Emergency1999, 2018-03-26:
Hello again,
now there is a different Problem:
I was able to find the data, but it is corrupted:
[image]
This is the message I get, when I try to open CURA.
Can someone please help me?
Thanks

[3.2.zip](https://github.com/Ultimaker/Cura/files/1848087/3.2.zip)

"Error Traceback" and "Logs" down below

**Error Traceback**
Traceback (most recent call last):
  File "C:\Program Files\Ultimaker Cura 3.2\plugins\LocalContainerProvider\LocalContainerProvider.py", line 117, in loadMetadata
    result_metadatas = clazz.deserializeMetadata(f.read(), container_id) #pylint: disable=no-member
  File "X:\3.2\build\inst\lib\python3.5\site-packages\UM\Settings\ContainerStack.py", line 412, in deserializeMetadata
  File "configparser.py", line 720, in read_string
  File "configparser.py", line 715, in read_file
  File "configparser.py", line 1077, in _read
configparser.MissingSectionHeaderError: File contains no section headers.
file: '<string>', line: 1
'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
  File "__startup__.py", line 12, in <module>
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
  File "Console.py", line 24, in <module>
  File "X:\3.2\build\inst\bin\cura_app.py", line 140, in <module>
  File "X:\3.2\build\inst\lib\python3.5\site-packages\UM\Application.py", line 338, in getInstance
  File "X:\3.2\build\inst\lib\python3.5\site-packages\cura\CuraApplication.py", line 315, in __init__
  File "X:\3.2\build\inst\lib\python3.5\site-packages\UM\Settings\ContainerRegistry.py", line 309, in loadAllMetadata
  File "C:\Program Files\Ultimaker Cura 3.2\plugins\LocalContainerProvider\LocalContainerProvider.py", line 123, in loadMetadata
    raise RuntimeError("Unable to deserialize metadata for container {filename}: {container_id}: {error_msg}".format(filename = filename, container_id = container_id, error_msg = str(e)))
RuntimeError: Unable to deserialize metadata for container C:\Users\Johannes\AppData\Roaming\cura\3.2\machine_instances\Ultimaker+3+%232.global.cfg: Ultimaker 3 #2: File contains no section headers.
file: '<string>', line: 1
'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\ [... cut ...]
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Current thread 0x000018f4 (most recent call first):
  File "X:\3.2\build\inst\lib\python3.5\site-packages\cura\CrashHandler.py", line 352 in _logInfoWidget
  File "X:\3.2\build\inst\lib\python3.5\site-packages\cura\CrashHandler.py", line 202 in _createDialog
  File "X:\3.2\build\inst\lib\python3.5\site-packages\cura\CrashHandler.py", line 80 in __init__
  File "X:\3.2\build\inst\bin\cura_app.py", line 113 in exceptHook

jackha, 2018-03-26:
Hi @Emergency1999 . We sometimes see that some files are "nulled", i.e. they contain only the \00\00\00... and unfortunately the file itself is then really corrupted and gone and the cause is still unknown. However if you upgraded from an old version, in some cases you could try to get some old files back:

Files are upgraded to the "new" cura, the original config files are backupped and they are still in the old config folder. Config folders are named after their Cura version.

Emergency1999, 2018-03-26:
Hi @jackha. Thanks for your awnser.
I was able to get an old file back to work. That's at least something.

Ghostkeeper, 2018-04-20:
Cura is at fault here for nulling these files. We've seen this before and back then the cause was multiple instances of Cura writing to a file at the same time. We should be careful to lock these files before writing to them, and I think that's missing here.

In an attempt to fix this, I've made one commit here: https://github.com/Ultimaker/Uranium/commit/1b08c567dbdfb0ab44a00d07795f48d4dd2aa0dc

Also Cura 3.3 will no longer crash for this error, but skip loading the corrupt file.

Ghostkeeper, 2018-05-01:
We still don't really know how to reproduce this bug though.

Zer0t3ch, 2018-05-02:
If there's issues reproducing the cause in order to diagnose and prevent, would it be possible to adjust the reaction to such an edge-case? A pop-up for the user to know what's going on or something would be wonderful.

Ghostkeeper, 2018-05-03:
Well, we don't want to always give a pop-up blindly, so we'd need to detect when concurrent access happens and that's what we can't reproduce. So no.

We have a better solution: A lock file that we check for existence before we're editing the files. If it exists, we know some other Cura instance is editing the files and wait for the lock file to disappear. We've used this for years while loading the settings ("Loading machines" phase of the loading screen). However I think the problem is that new features have been added that manipulate files in the file system and those don't check the lock file.

Treeed, 2018-05-03:
Okay, I think I had that problem a LOT of times, but right now is the first time I got a popup actually telling me that something went wrong rather than just quietly resetting all my settings (Same popup as above). 
I'm not sure if previous instances of (what I believe was) this problem were after upgrades too. I didn't google for those as there was no error to google.
This time was after upgrading from 3.2.1 to 3.3.1 on the second start of 3.3.1. All settings in "definition changes" "extruders" "machine_instances"  and "user" were filled completely with zeros.
My upgrade procedure leading to the error was as follows: 

> 3.2.1 Shows the little box at the bottom telling me there's a new version
> I click the download button and the website opens in Firefox in the foreground
> I download and start the installation package
> The package asks me whether I want to uninstall the old cura version
> I select "yes"
> The uninstallation procedure finishes successfully
> The installatopn precedure finishes successfully and asks me whether I want to start cura
> I notice cura is still running in the background
> I close cura as I don't want the old version running any more. I don't care whether the uninstallation failed silently as long as the new version runs
> I click finish in the setup with the start cura checkmark set
> Cura starts without a problem
> I start setting up my model and change a few print settings. No cura settings though
> I want to start printing but notice none of my USB ports recognize new devices
> I close cura using Alt+F4 (mapped on my mouse)
> I restart the computer
> I get error messages trying to start cura; my settings are filled with zeros

I tried to describe this as detailed as possible in case a small step is relevant as I currently have no time to try and reproduce the problem myself.

jackha, 2018-05-07:
@Treeed thank you for the detailed explanation. When you got the error for the second time, did the file also contain all zeros? We're still investigating the zeros, but we still don't know for sure where it comes from. Also, do you use Windows or a different OS?

Treeed, 2018-05-07:
@jackha No problem, thanks for your work! I'm pretty sure it did, but I'm not entirely sure. I'll report back if it happens again.
I use a very broken Windows 7. One of its problems is, that i have to cut power to shut it down because it gets stuck on the "logging off" screen, so it sort of crashes on every shutdown. Cura however is always closed before I shut down.

jackha, 2018-05-30:
After beta release we're fixing the top xxx crashes. If this is one of them, then we'll fix it. Currently we already some effort in it, but without finding something useful yet.

jackha, 2018-05-30:
So closing it for now.

closed 2018-05-30 by jackha