Ultimaker/Cura #3562: [3.2.1] Cura crashes with all temperatures = 0

mtmello2003, 2018-03-26:
Win10 x64

If you set all temperatures parameters with 0, cura crash when try to slice.

Expected behavior: create a gcode without any M140, M109, M104 temperatures command.

jackha, 2018-03-26:
I've set all temperatures to 0 and I think you mean that CuraEngine crashes? It returned a -6 and Cura says "Slicing...". Cura itself runs fine though.

mtmello2003, 2018-03-26:
Hmm... I´m using dual extruder config and try to set the another extruder with all temp = 0 and curaEngine crash when slicing.

If you can´t reproduce this error, I can export the project.

jackha, 2018-03-27:
I just retested it on our 3.3 branch and CuraEngine doesn't seem to crash anymore. If you can export the project I can retest with that.

ianpaschal, 2018-03-27:
Just tested on our 3.3 beta build and it's fixed. Yay!

closed 2018-03-27 by ianpaschal