Ultimaker/Cura #3563: Fails to display second extruder's temperature

kirilledelman, 2018-03-26:
I've a modified Prusa Mk3 with 2 extruders. Since it's not in the list of printers, I've added it as Custom FDM printer with 2 extruders. The Monitor tab shows bed temperature and first extruder's temperature correctly, but not the second. Digging around, I found in Cura's log the following error message:
```
Could not parse hotend temperatures from response: b'ok T:11.56 /0.00 B:12.00 /0.00 T0:11.56 /0.00 T1:12.00 /0.00 @:0 B@:0 @0:0 @1:0\n'
```
in file/function USBPrinting.USBPrinterOutputDevice._listen

I've added some exception logging code and see it's triggering `list index out of range` in the code where it's looping over all matches to "T(\d*): ?([\d\.]+) ?\/?([\d\.]+)?" regular expression.

It looks to me like, it's finding my printer's `T0: temp/targetTemp` and calls `self._setHotendTemperature( 0, temp )` fine for first extruder, but calling it for the second triggers this exception. It's as if internally, inside `self._setHotendTemperature` the number of extruders is still 1, and it's hitting some array's boundaries.

I was unable to rebuild Cura on MacOS (using cura-build) packages myself. Dependencies hell. But I can hack the py scripts sitting inside app bundle.

Anyone has any ideas what else I can try?

kirilledelman, 2018-03-26:
@slundell, @Ghostkeeper, do you guys have any ideas? 

I know I am running the last public release (3.2.1), not the down-to-the-wire version, but maybe you have some insight as to why `_setHotendTemperature` call with `1` for extruder, for an FDM printer with 2 extruders results in `list index out of range` exception.

I tried replacing just the USBPrinting plugin folder in my Cura with the latest, but of course, the API has changed and it didn't work. I'm so far unable to compile Cura myself on a Mac, stuck on some dependency, and running it on a Mac from source results in exceptions as well.

ianpaschal, 2018-03-28:
Possibly related to #3311? Basing this on the association between 3rd party, usb printing, and temperature readout.

GMagician, 2018-03-28:
This issue is also reported here: #3361

fieldOfView, 2018-03-28:
Note to self:
https://github.com/Ultimaker/Cura/issues/3361#issuecomment-367131150

ianpaschal, 2018-03-28:
Yeah marking as duplicate and closing.

kirilledelman, 2018-03-28:
Same thing happens in 3.3 beta.

kirilledelman, 2018-03-29:
Does anyone know if the print would still start and work correctly, even though the temperature doesn't display? Would second extruder preheat?

Ghostkeeper, 2018-04-20:
The error shown in your log shouldn't block you from USB printing. I'd expect the print to start and work correctly.

ChrisTerBeke, 2018-05-24:
Fixed by https://github.com/Ultimaker/Cura/pull/3597.

closed 2018-05-24 by ChrisTerBeke