Ultimaker/Cura #3564: master branch - some settings changes are reset to original value

smartavionics, 2018-03-26:
**Application Version**

master branch after commit d360c6d78

**Platform**

Linux

**Qt**

5.8.0

**PyQt**

5.8.2

**Display Driver**

**Printer**

Kossel Mini

**Steps to Reproduce**

Change a setting such as Printing Temperature and slice model.

**Actual Results**

After slice has finished, the setting jumps back to the original value and the new values was not used
for the slice.

**Expected results**

The new setting value should not revert back and the slice should use the new setting value.

**Additional Information**

This behaviour has started since commit d360c6d78. If I can narrow it down further I will let you know.

smartavionics, 2018-03-26:
Could it possibly be caused by https://github.com/Ultimaker/Uranium/commit/d9c4e7423036affb8ce5ebfeeb0d2671d05cb02a ?

smartavionics, 2018-03-26:
Could it possibly be related to https://github.com/Ultimaker/Uranium/commit/d9c4e7423036affb8ce5ebfeeb0d2671d05cb02a ?

smartavionics, 2018-03-27:
I am getting nowhere with this, mainly due to my ignorance of how the whole UI functions. Why should the rate at which digits are entered into a setting field influence the behaviour? Anyone got any ideas?

smartavionics, 2018-04-06:
This appears to have been fixed today, closing...

closed 2018-04-06 by smartavionics

Ghostkeeper, 2018-04-30:
Yeah this was a duplicate of https://github.com/Ultimaker/Cura/issues/3590 .