Ultimaker/Cura #3566: Crash on 3.2.1 - loading or saving

JamesRimot, 2018-03-26:
Hi

Have tried the trick of deleting the appdata/roaming/cura/3.2 folders and reinstalling.  Rebooted.  Keeps crashing.

I had been able to create .gcode with a sample STL file.  But now it consistently crashes with any I/O.

I don't see any dump file to attach.  

[test-output-blank.curaproject.zip](https://github.com/Ultimaker/Cura/files/1849246/test-output-blank.curaproject.zip)

**Application Version**
3.2.1

**Platform**
Windows 10 64 bit

**Qt**
ultimaker

**PyQt**
ultimaker

**Display Driver**
AMD Radeon R7 21.19.128.4

**Printer**
monoprice 3D mini V2

**Steps to Reproduce**
deleted /appdata/roaming/cura/3.2 folder
clean install of Cura. 
Reboot
 Open Cura. 

**Actual Results**
 Open file.  crash.
or   save file.  crash

**Expected results**
 Open file.  crash.
or   save file.  crash

**Additional Information**
n/a

fieldOfView, 2018-03-26:
Does dragging a file into the Cura window work? It might be “just” the file dialogs acting up.

JamesRimot, 2018-03-27:
Hi FieldOfView
That actually worked!
I think the file dialogs are acting up.
Is there a way to remedy the file dialogs causing crashes?  
Thanks again - Cura is great other than this crashing issue.
Cheers,
James

closed 2018-03-27 by JamesRimot