Ultimaker/Cura #3567: Need to have Skirt, Brim and Raft in Build Plate Adhesion settings!

adamantiz, 2018-03-26:
Hello,

This software is wasting a lot of my time, electricity and plastic now.

My printer takes a bit of driving around before the plastic starts to stick to the build plate.
I believe this may be due to the fall in temperature when it first begins extruding (due to cool plastic).

I NEED the engine to have options for both skirt and raft, or skirt and brim, or all. So that I can remove the non-sticking plastic, rather than having to stop the whole print, clean and start again until something sticks.

If I need a raft layer, then I also need a skirt to peel away. Because if the outer brim of the raft is removed, then the in-fill of the raft peels back with every line, by at least enough to ruin the raft, therefore ruin the print.

I seriously do not understand why you have combined these options and set them to alternate. All three options should be allowed.

Also, the raft layer needs different options. Just 2 layers is enough to prime the build plate. There is no need to print 5 layers of a raft. The current settings only allow it to be as thin as 3 or 4 layers. What a total waste!

Please, amend the settings (rather than tell me I can edit things myself, as was the response to my last ticket).

Thanks,
Adam.

fieldOfView, 2018-03-26:
Shall we send someone over to do your dishes too?

adamantiz, 2018-03-26:
ok, so if I take the source code and make this change, would my update be implemented?

I am not going to waste my time in visual studio, learning this.program's code if not...

fieldOfView, 2018-03-26:
I'm not affiliated with Ultimaker, so I have no say in what goes in. But code that is well written and benefits a group of user is generally accepted. 

Most of the work would be in https://github.com/Ultimaker/CuraEngine. I have no experience with setting up Visual Studio to build CuraEngine. Perhaps @smartavionics can chime in about how complex the changes would be. He's also not affiliated with Ultimaker, but knows a LOT about the engine.

For reference, see https://github.com/Ultimaker/Cura/issues/942 for a previous request of the same functionality.

fieldOfView, 2018-03-26:
Visual Studio is not great for cross platform development. Cmake and friends work on Linux, OSX and Windows. If you can create project files for Visual Studio that build and run the project reliably, I think a PR would be appreciated.

smartavionics, 2018-03-26:
Hi. Sorry, I know almost zilch about building applications for Windows as I only use Linux.

As @fieldOfView says, code that is well written (you know, like all of mine, cough, cough) is likely to be incorporated if it doesn't detract from existing functionality and isn't so obscure in function that only one person on the planet would find it useful.

And, of course, at the end of the day, Cura is open source so if you want it to do something and the Cura guardians don't give it their blessing, you can always just use it yourself. I have a growing collection of extremely useful and good quality PRs that the Cura devs have turned down and have long abandoned the hope of having all of my gems incorporated in "standard" Cura. Good luck!

adamantiz, 2018-03-26:
lol, visual studio is the goto solution FOR cross platform development. Not the place to debate it I know, but in my honest opinion, any programmer using anything other than visual studio is wasting time. They bought out Xamarin years ago... I wouldn't even develop for a mac in anything other than visual studio.

If I use the Unity3D game engine, I am using visual studio...
If I make a website with a login, Visual Studio...
A desktop app, visual studio,
A mobile (android) app, visual studio.
Python or other languages, Visual Studio...

It is a surprise to me, than this program, using C++ is not a visual studio project.

I have a fully licenced copy of Visual Studio Enterprise edition from 2015. I got it for FREE as a student.
Shortly afterwards they made it a subscription based application, and gave a free version. Anyone looking to start programming need not look anywhere else.

It hinders this projects development, because... well, I for one won't have enough time to look at the code, or make edits, without visual studio. I am sure others have the same opinion.

Thanks.

smartavionics, 2018-03-26:
You will have to forgive my ignorance. Does VS support building C++ programs for Linux and Mac?

fieldOfView, 2018-03-26:
And more importantly, does VS support building C++ programs _on_ Linux and Mac? Most Cura development happens on flavors of Linux.

adamantiz, 2018-03-26:
ok, I will totally derail the thread to make a point...

Do they teach students in college and university Linux?
Do they use Linux in any academic institutions? Not at lower levels at least...

Does Microsoft give software to students for free, including Windows and VS and even Office? YES!
Is it easy enough to find a crack and install any software which you cannot afford? yep! :)
So are new developers using Linux, or Microsoft products?

Yes you can develop for a mac using visual studio and the xamerin plugin.

I can't understand why you are using Linux either, unless you are using it to run a server of some kind... or for hacking. I am sure you have heard of virtual box... install Windows on Linux or duel boot... and get visual studio, so that others can contribute to the open source project.

If you are making a project, which specifically demands developers install Linux, then you are not going to get much help with it from newer developers. Die-hard Linux fans, of which there are not many claim Linux to be the best thing, but it is the most restricted and restricting, when it comes to developments, because people have to go out of their way to learn how to use it, while everyone can use Microsoft products, for free!

closed 2018-03-26 by adamantiz

Patola, 2018-03-26:
Do you need to sh*tpost like that to make your point?

First the brim/skirt thing. Now you have to insult all people who use other platforms and require that everyone use your preferred proprietary solution. I don't think you're out to earn sympathy, amirite?

Cross-development platform is useful because many volunteers use Mac OS X and Linux. You can see it as 'diversity' and 'inclusion', only geared towards OSs. And I thank Ultimaker for that, because that is exactly the reason I (and other contributors) make our patches. Cura would be much poorer if it wasn't for its cross-platform development, I guess.

adamantiz, 2018-03-26:
you mistake advice for demands... and use an abusive tone... you insinuate I was looking for sympathy...
No, you are not right.

Your idea of cross platform development seems as limited as the platforms you choose to don.
Because it excludes developments from the biggest, most widely used platform. Windows!

There was no need for your tone, and your post equated to a sh*t post to M8.

adamantiz, 2018-03-26:
closed to prevent trolls

ianpaschal, 2018-03-27:
@adamantiz:

@fieldOfView is right; it's simply not a big priority for us. This is FOSS though so you are welcome to submit a pull request. We do review everything, so if the code is high quality, we will merge it. You'll have to be open to making requested changes though.

As for the rest of this thread, I appreciate you closing the topic to prevent trolls but it's a little late thanks to your own actions. I think about 20% of the Cura team actually uses Windows for development and by no measure is VS the best cross platform development program. Comments like,

>/facepalm at this not being a visual studio project file...

come across as, "Oh my god, everyone is so dumb" and easily rubs people the wrong way. Combine it with a tirade about how a piece of proprietary software that none of us is obviously the best way to produce FOSS and of course you get a bad reaction.

I don't want to dissuade you from contributing to Cura as the whole point re: VS is that we would like everyone who is remotely interested to be involved in improving the software, but please think about how your tone comes across others.

Looking forward to your PR. 😉

adamantiz, 2018-03-27:
I /facepalm because it means, if it is not a VS project, I won't have time to contribute.
It seems like it would be a very simple and 2minute job, for anyone who is already familiar with the code.

So it wasn't negativity on my part to be fair, it was negativity in the perception/perspective of the reader.

Thank you for putting it into context.

Ghostkeeper, 2018-04-20:
You can generate a VS project for CuraEngine with the following command ran from CuraEngine's source directory, if VS is your preference:

    cmake -G "Visual Studio 10"

Or whichever version of Visual Studio you're using. That should get you started.